modelica/ModelicaStandardLibrary #3435: Analysis of example models of ModelicaTest v4.0.0-beta.1 failing a pedantic translation

beutlich, 2020-02-23:
The following models/blocks with StopTime annotation fail translation if pedantic=true:

- [ ] Modelica.Blocks.Examples.PID_Controller
- [ ] Modelica.Clocked.Examples.Systems.EngineThrottleControl
- [ ] Modelica.Clocked.Examples.Elementary.ClockSignals.RotationalSample
- [ ] Modelica.Clocked.Examples.Elementary.ClockSignals.LogicalSample
- [ ] Modelica.Electrical.Spice3.Examples.Oscillator
- [ ] Modelica.Electrical.Spice3.Examples.CoupledInductors
- [ ] Modelica.Magnetic.FluxTubes.Examples.Hysteresis.HysteresisModelComparison
- [ ] Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis1
- [ ] Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis2
- [ ] Modelica.Magnetic.FluxTubes.Examples.Hysteresis.ThreePhaseTransformerWithRectifier
- [ ] Modelica.Magnetic.QuasiStatic.FluxTubes.Examples.Leakage.CylinderLeakage
- [ ] Modelica.Magnetic.QuasiStatic.FluxTubes.Examples.Leakage.GeneralLeakage
- [ ] Modelica.Fluid.Examples.HeatingSystem
- [ ] Modelica.Fluid.Examples.AST_BatchPlant.Test.OneTank
- [ ] Modelica.Fluid.Examples.NonCircularPipes
- [ ] Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls
- [ ] Modelica.Fluid.Examples.InverseParameterization
- [ ] Modelica.Fluid.Examples.Explanatory.MeasuringTemperature

beutlich, 2020-02-23:
Hysterisis models are addressed by #3427.

beutlich, 2020-02-23:
Spice3 models are addressed by #3438.

casella, 2020-03-03:
@beutlich, I just checked all the Modelica.Fluid examples on branch v4.0.0-beta.1 using Dymola 2020x and pedantic mode, but I don't get any error or warning. Can you please report which errors do you get?

beutlich, 2020-03-07:
Here the warnings:

```
Translation of Modelica.Fluid.Examples.HeatingSystem:

Some variables are iteration variables of the initialization problem:
but they are not given any explicit start values. Zero will be used.

Iteration variables:
der(tank.V)
```

```
Translation of Modelica.Fluid.Examples.AST_BatchPlant.Test.OneTank:

The initial conditions are not fully specified.
Dymola has selected default initial conditions.

Assuming fixed start value for the continuous states:
tank.level(start = tank.level_start)
tank.medium.T(start = tank.T_start)
```

```
Translation of Modelica.Fluid.Examples.NonCircularPipes:

Redundant consistent initial conditions.

Removed the following equations which are redundant and consistent: 
circular_pipe.port_a.p = circular_pipe.mediums[1].p;
annulus_pipe.port_a.p = annulus_pipe.mediums[1].p;

This required evaluation of 
the interactive parameters 
boundary.p(start = 1000000.0)

The redundancies involve
initial equations: 
circular_pipe.mediums[1].p = 1000000.0;
annulus_pipe.mediums[1].p = 1000000.0;
```

```
Translation of Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls:

The initial conditions are not fully specified.
Dymola has selected default initial conditions.

Assuming fixed start value for the continuous states:
PID.I.y(start = PID.I.y_start)
```

```
Translation of Modelica.Fluid.Examples.Explanatory.MeasuringTemperature:

Some variables are iteration variables of the initialization problem:
but they are not given any explicit start values. Zero will be used.

Iteration variables:
der(openTankCold3.m)
der(openTankHot1.m)
der(openTankHot2.m)
```

I can no longer reproduce Modelica.Fluid.Examples.InverseParameterization.

beutlich, 2020-03-16:
@AHaumer @christiankral Can you please have a look at Modelica.Blocks.Examples.PID_Controller. It is not a big deal to set this initial condition (somewhere hidden in the component), the actual trouble is, if PID is missing such an option generally.

christiankral, 2020-03-16:
> @AHaumer @christiankral Can you please have a look at Modelica.Blocks.Examples.PID_Controller. It is not a big deal to set this initial condition (somewhere hidden in the component), the actual trouble is, if PID is missing such an option generally.

@beutlich So this means that we actually need a new ticket to consistently set the initial conditions in each case of controller configurations (P, PI, PD, PID).

beutlich, 2020-03-16:
Yes, that is the question I was asking myself.

beutlich, 2020-04-27:
Removing milestone after discussion stopped.

beutlich, 2020-05-15:
Modelica.Fluid.Examples.TraceSubstances.RoomCO2WithControls was fixed en passant by a0b60a045ba45bdf0bd6e11573a791cca958d3ac.

beutlich, 2021-04-10:
Closing after MSL v4.0.0 was released. It is a "WontFix" for the four Fluid examples that still raise initialization warnings in pedantic mode.

- [ ] Modelica.Fluid.Examples.HeatingSystem
- [ ] Modelica.Fluid.Examples.AST_BatchPlant.Test.OneTank
- [ ] Modelica.Fluid.Examples.NonCircularPipes
- [ ] Modelica.Fluid.Examples.Explanatory.MeasuringTemperature

closed 2021-04-10 by beutlich