modelica/ModelicaStandardLibrary #3445: Wrong rotation direction in planaRotation functions

dariomangoni, 2020-02-24:
`Modelica.Mechanics.MultiBody.Frames.planarRotation` and `Modelica.Mechanics.MultiBody.Frames.TransformationMatrices.planarRotation` either rotate in the opposite direction (respect to right-handed standard) or the documentation is quite unclear.

Given:
- rotation axis along +Z
- rotation angle of 90deg
- a starting vector `v1` along +X

I would expect to get a rotation matrix that rotates the vector `v1` according to `R*v1` in the +Y direction, not in -Y direction as it currently happens.

The expected R for me should have been
`[  0.0, -1.0,  0.0;  
  +1.0,  0.0,  0.0;  
    0.0,  0.0,  1.0]`
instead is  
`[  0.0, +1.0,  0.0;  
  -1.0,    0.0,  0.0;  
    0.0,   0.0,  1.0]`

Am I missing something?

tobolar, 2020-03-02:
`Modelica.Mechanics.MultiBody.Frames` defines functions to perform all common operations on (position, etc.) vectors and tensors needed for a multi-body models. Whatever you intend to do, please use consequently functions included in this package. The functions shall be sufficiently documented.
I suppose you intend to do something like:
```
v1 = {1,0,0};  
R = planarRotation({0,0,1}, from_deg(90), {0,0,0});
v2 = resolve2(R, v1);
```
This shall, actually, work. Please try.

dariomangoni, 2020-03-02:
Exactly. Now, what do you expect from the rotation example that you gave me?
Do you expect `v2={0, 1, 0}` or `v2={0,-1,0}`?
Because I think you'll be surprised about the result.

tobolar, 2020-03-02:
Well I would expect `v2={0,-1,0}`. And this is what actually I get.
```
function testPlanarRotation
  output Real v2[3];
protected 
  Real v1[3] = {1,0,0};
  Modelica.Mechanics.MultiBody.Frames.Orientation R;
algorithm 
  R := Modelica.Mechanics.MultiBody.Frames.planarRotation(
    {0,0,1}, Modelica.SIunits.Conversions.from_deg(90), 0);
  v2 := Modelica.Mechanics.MultiBody.Frames.resolve2(R, v1);
end testPlanarRotation;
```
Dymola: `//  = {6.123233995736766E-17, -1.0, 0.0}`

But maybe you can better specify what part of documentation is not clear to you. I could try to improve it.

HansOlsson, 2020-03-03:
I believe that the issue is that "rotating a vector 90 degrees" as @dariomangoni wants is simply not the mind-set of MultiBody libraries. 

Instead the mind-set is that vectors can be transformed between different reference frames.

The function resolve1 more closely corresponds to rotating a vector in that sense (it takes a vector in a rotated reference frame and returns it in the non-rotated reference frame) whereas resolve2 goes in the other direction .

dariomangoni, 2020-03-03:
Yes, exactly what @HansOlsson said.
But I think it shouldn't be just a "mindset", but a proper description in the documentation.

From the documentation I read that planarRotation returns an  
"orientation object R that describes the orientation to **rotate in the plane** along unit axis e from frame 1 into frame 2"  
then by doing R*v1 I would expect that v1 is rotated as the planarRotation argument said.

In my opinion the documentation should say  

"The function call Frames.planarRotation(e, angle, der_angle) returns an orientation object R ~~that describes the orientation to rotate~~ _rotated in the plane_ along unit axis e from frame 1 into frame 2"  

since the orientation object R **does not** rotate in the plane (otherwise I would expect R*v1 give {0,+1,0}). On the contrary, **is** rotated in the plane. But this is my point of view.
Do as you please! Thanks again

PR #4217 (2024-01-09): Documentation: clarify mind-set of planar rotation
Closes #3445
closed 2024-01-09 by hubertus65