modelica/ModelicaStandardLibrary #3446: External functions should specify 'Include' to make use of existing prototypes

henrikt-ma, 2020-02-25:
The external function `Modelica.Utilities.Files.fullPathName` seems to be typical in the sense that it doesn't use the `Include` annotation, meaning that the function call isn't checked against the existing prototype in [ModelicaInternal.h](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Resources/C-Sources/ModelicaInternal.h).

It would improve the quality of MSL if all external functions made use of the `Include` annotation.

More generally, the use of `Include` has a clear benefit in tools that call the external function from C++ code, as the presence of `Include` forbids the tool from generating a prototype based on how the function is called, meaning that the compiler will refuse to compile the call instead of compiling the call based on an incorrect automatically generated prototype (which will then likely cause a simulation crash).  In tools that call the external function from C code, there is no drawback of specifying `Include`, but the advantage is smaller as the compiler will still compile the call with a warning if the prototype is missing.

beutlich, 2020-02-25:
I believe we discussed this topic already some time ago, but I cannot find the thread. Maybe it was a private mail discussion only or I my memory might even be wrong.

henrikt-ma, 2020-02-26:
If not even you can find it, @beutlich, then it isn't in the archives. :)

PR #3453 (2020-02-28): Add Include annotation for external function interface
Resolves #3446 as proposed.
closed 2020-02-28 by beutlich