modelica/ModelicaStandardLibrary #3455: Check of conversions

HansOlsson, 2020-02-27:
Continuing 
https://github.com/modelica/ModelicaStandardLibrary/issues/3181#issuecomment-549259166

Old classes missing (note, I had https://github.com/modelica/ModelicaStandardLibrary/pull/3453 checked out, but I don't think that matters):
- [ ] Modelica.Magnetic.FundamentalWave.BaseClasses.Machine
- [ ] Modelica.Magnetic.FundamentalWave.BasicMachines.InductionsMachines.{Im_SlipRing,IM_SquirrelCage}
- [ ] Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousMachines.{SM_ElectricalExcited,SM_PermanentMagnet,SM_ReluctanceRotor}
- [ ] Modelica_Synchronous.RealSignals.TimeBasedSources.Sin (for conversion of f->freqHz) (should be Sine)

I also tried to check that all public elements in MSL are converted.

That seems to be the case for classes in MSL (excluding Examples, UserGuide, and partial ones).

For components conversions are missing for some "internal" variables, e.g., vbc in PNP due to https://github.com/modelica/ModelicaStandardLibrary/pull/3178
but for parameters and graphical ones only the following seems to lack conversions:
- Modelica.Electrical.Machines.Utilities.SwitchYD idealCommutingSwitch
- Modelica.Electrical.Spice3.Internal.MOS2 vp
- Modelica.Magnetic.QuasiStatic.FundamentalWave.Utilities.SwitchYD idealCommutingSwitch
- Modelica.Mechanics.MultiBody.Parts.Body z_a_start
- Modelica.Fluid.Machines.BaseClasses.PartialPump show_NPSHa
I'm not sure if any of them are relevant.

(The above excluding classes that are partially converted.)

HansOlsson, 2020-02-27:
**Including** the partially converted ones we find some more classes that are gone without conversions (as far as I see).
- Modelica.Electrical.QuasiStationary.SinglePhase.Interfaces.RelativeSensor
- Modelica.Electrical.QuasiStationary.MultiPhase.Blocks.SingleToMultiPhase
- Modelica.Electrical.QuasiStationary.MultiPhase.Interfaces.RelativeSensor

Except for the previously reported parameter there seemed to be no issues for component.

beutlich, 2020-02-28:
>     * Modelica.Magnetic.FundamentalWave.BaseClasses.Machine

Hm, I don't understand. There is

https://github.com/modelica/ModelicaStandardLibrary/blob/e40361763bdf818b9d6bbeaf8bf1f28212b3c80c/Modelica/Resources/Scripts/Conversion/ConvertModelica_from_3.2.3_to_4.0.0.mos#L224-L225

where Modelica.Magnetic.FundamentalWave.Interfaces.PartialBasicInductionMachine is available in MSL 3.2.3 and Modelica.Magnetic.FundamentalWave.BaseClasses.Machine is available in MSL 4.0.0 Beta.1.

beutlich, 2020-02-28:
> * Modelica.Electrical.Spice3.Internal.MOS2 vp
> * Modelica.Mechanics.MultiBody.Parts.Body z_a_start
> * Modelica.Fluid.Machines.BaseClasses.PartialPump show_NPSHa

These removals are already documented as non-backward compatible changes in the edited release notes (Modelica.UsersGuide.ReleaseNotes.Version_4_0_0).

beutlich, 2020-02-28:
> * Modelica.Magnetic.FundamentalWave.BasicMachines.InductionsMachines.{Im_SlipRing,IM_SquirrelCage}

Covered by https://github.com/modelica/ModelicaStandardLibrary/blob/e40361763bdf818b9d6bbeaf8bf1f28212b3c80c/Modelica/Resources/Scripts/Conversion/ConvertModelica_from_3.2.3_to_4.0.0.mos#L164-L167

> * Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousMachines.{SM_ElectricalExcited,SM_PermanentMagnet,SM_ReluctanceRotor}

Covered by https://github.com/modelica/ModelicaStandardLibrary/blob/e40361763bdf818b9d6bbeaf8bf1f28212b3c80c/Modelica/Resources/Scripts/Conversion/ConvertModelica_from_3.2.3_to_4.0.0.mos#L168-L169

christiankral, 2020-02-29:
> Hm, I don't understand. There is [...]

@beutlich May I ask what part you do not understand here? I checked the conversion based on `ModelicaTestConversion4.Magnetic.FundamentalWave.Issue496.PartialBasicInductionMachine` and it works without issues.

beutlich, 2020-02-29:
I do not understand what shall be fixed here. Or in other words: I do not understand, why @HansOlsson reported it as a potential conversion issue.

christiankral, 2020-02-29:
OK then. I came to the same conclusion as @beutlich 

- [x] Modelica.Magnetic.FundamentalWave.BaseClasses.Machine
- [x] Modelica.Magnetic.FundamentalWave.BasicMachines.InductionMachines.{IM_SlipRing,IM_SquirrelCage}
- [x] Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousMachines.{SM_ElectricalExcited,SM_PermanentMagnet,SM_ReluctanceRotor}

convert all good and correctly.

christiankral, 2020-02-29:
> **Including** the partially converted ones we find some more classes that are gone without conversions (as far as I see).
>
>     * [...]

These issue shall be covered by 9d3a389edd5c55a52946df40cd6d307706f0ae99 and d9ecb20f4b2d4032bb516971f1c44f7459d32cb6

beutlich, 2020-02-29:
PR would be welcome.

christiankral, 2020-02-29:
>     * [ ]  Modelica.Electrical.Machines.Utilities.SwitchYD idealCommutingSwitch
> 
>     * [ ]  Modelica.Magnetic.QuasiStatic.FundamentalWave.Utilities.SwitchYD idealCommutingSwitch

These two issues is taken care of in 3fc099c829db57dc0fcf9d3459dd330ccaba1204 and #3467, respectively

HansOlsson, 2020-03-02:
> Hm, I don't understand. There is [...]

Hmm... 

I had to check more as well, and the issue is another line:
https://github.com/modelica/ModelicaStandardLibrary/blob/e40361763bdf818b9d6bbeaf8bf1f28212b3c80c/Modelica/Resources/Scripts/Conversion/ConvertModelica_from_3.2.3_to_4.0.0.mos#L1821-L1821
that tries to convert something in `Modelica.Magnetic.FundamentalWave.BaseClasses.Machine`; but as noted that class was introduced in 4.0.0.

HansOlsson, 2020-03-02:
> These removals are already documented as non-backward compatible changes in the edited release notes (Modelica.UsersGuide.ReleaseNotes.Version_4_0_0).

Ok, I guess I should have looked there first.
However, I noticed that this implies that something like convertMessage for components would be a good idea in the future.

HansOlsson, 2020-03-10:
(In addition to #3492) I found the following additional issues:

- [ ] Modelica.Electrical.QuasiStationary.SinglePhase.Sensors.CurrentSensor had arg_i instead of arg_v as new connector name in conversion script.
 -[ ] Modelica.Media.Air.MoistAir.specificInternalEnergy_pTX had conversion for R->R_s
 -[ ] Modelica.Media.Air.ReferenceAir.Air_Utilities.Basic, Modelica.Media.Air.ReferenceMoistAir.Utilities.Ice09_Utilities.Basic, and Modelica.Media.Air.ReferenceMoistAir.Utilities.Water95_Utilities does not have R directly but in Constants

PR #3475 (2020-03-10): refs #3455: Fix conversion
Remove conversion from new class.

Is there anything more to fix? Was not that obvious for me.

Fix #3455.
closed 2020-03-10 by beutlich