modelica/ModelicaStandardLibrary #3472: Remove of reference signals of zero sequence currents of electric machine examples

christiankral, 2020-03-01:
The zero sequence currents of the electric machine simulation models shall be zero in most cases. For these cases the zero sequence currents shall be removed from the list of reference signals. Otherwise we have to manually validate upcoming regressions as in #3412 over and over again.

AHaumer, 2020-03-11:
I fully agree with @christiankral:
In most cases the zero sequence currents are zero, the results are numerical noise. They are no appropriate signals for reference results. It would be much better to compare one of the phase currents,

christiankral, 2020-07-02:
Refs  #3472, #3497, #3587, #3591, https://github.com/modelica/MAP-LIB_ReferenceResults/pull/3

closed 2020-07-02 by christiankral