modelica/ModelicaStandardLibrary #3479: Modelica.Clocked.Examples.Systems.Utilities.ComponentsThrottleControl.Engine uses shiftSample with parametric resolution on an event clock

carlj-w, 2020-03-03:
_Modelica.Clocked.Examples.Systems.Utilities.ComponentsThrottleControl.Engine_ has an event clock (_engine.crankshaftPositionEvent_) defined in the component _engine_. This clock is used as input to _engine.inductionToPowerDelay_, where it defines the clock of _engine.inductionToPowerDelay.delay_, which is an instance of _Modelica.Clocked.RealSignals.NonPeriodic.FractionalDelay_.

Now, _FractionalDelay_ uses a shiftSample with resolution, using the resolution set by the parameter _resolution_. A shiftSample can only be applied to an event clock if the resolution is set to 1 (See Modelica specification 3.4, page 206). In this case, the parameter _resolution_ is set to 1, but it is not indicated that the parameter should be evaluated during translation with the `Evalute = true` annotation.

It also seems to me that the parameter _resolution_ should have `Evalute = true` since it is used to determine the size of the array _u_buffer_ (the same goes for the parameter _shift_).

christoff-buerger, 2020-03-05:
Just a short feedback: I have this on my radar and will look into it next Monday/Tuesday.

beutlich, 2020-03-10:
@christoff-buerger Can you please estimate how severe this issue is w.r.t. the Beta.2 tag of the MSL v4.0.0?

christoff-buerger, 2020-03-17:
@carlj-w: I do not think that `Evalute = true` annotations are the way to go. The reason is simple: `RealSignals.NonPeriodic.FractionalDelay` components can be used also in non-event-clock based partitions, where the resolution very well can be different from 1.

Instead, I think the model can be made safe by _forceful_ preparation of the `Examples.Systems.Utilities.ComponentsThrottleControl.InductionToPowerDelay` model for usage in the event-clock partition given by the observed crank-shaft it is _deliberately_ modeled for. The respective intention can be denoted by doing a `final` modification of the `resolution` of its `delay` component to 1, i.e.,
```
RealSignals.NonPeriodic.FractionalDelay delay(
    shift = 1,
    final resolution = 1)
```

**Regarding the other parametric expressions used to define multi-dimensional component sizes:** I hesitate to mark each and every parameter used to define a vector, matrix etc size with `Evalute = true`; this is spreading like cancer and unnecessary. It is the responsibility of the Modelica tooling you are using to satisfy the standard and figure out that parameters are used to define multi-dimensional sizes; in particular, your tooling should be clever enough to know which information it needs to statically evaluate at compile time to be capable to generate code such that it can simulate your standard conform model. It is **not** a modeler's responsibility to help a tool generate code.

PR #3508 (2020-03-20): #3479 fix
Fix #3479; please check the issue for a rationale of the final modification.
commit 6a78e727 (2020-03-20): Merge pull request #3508 from christoff-buerger/#3479-fix

#3479 fix
closed 2020-03-20 by beutlich