modelica/ModelicaStandardLibrary #3486: Modelica.Utilities.Files.loadResource has changed?

AHaumer, 2020-03-05:
I'm just testing a few of my libraries with MSL 4, and I found the following:
`    Modelica.Utilities.Files.loadResource("modelica://EMOTH/Resources/DrivingCycles/WLTC.txt"),`
worked previously fine. Now with MSL 4 the absolute path name is returned wrong:
C://work/modelica://EMOTH/Resources/DrivingCycles/WLTC.txt
i.e. withput removing modelica://

beutlich, 2020-03-05:
From a mail sent by @HansOlsson previously to some members by mail:

Loading the correct ModelicaServices means:

* Load MSL

* Ensure that Tools>Options>Package Browser: Show protected classes is active (in Dymola 2020 that is Edit>Options> Package Browser: Show protected).

* ModelicaServices (if loaded), context menu>UnLoad (in rare cases you might have to do this twice)

* Open ModelicaServices from c:\program files\Dymola 2020x\Modelica\Library\ModelicaServices 3.2.3

* Ignore warnings about incompatible version.

Best Regards,
Hans OLSSON

christoff-buerger, 2020-03-05:
@AHaumer: Like @beutlich said, the issue is likely that you loaded the default MSL Modelica Services, which is just a mockup/dummy each tool must implement itself.

Besides file handling etc (`Modelica.Utilities.*`) also animations are not implemented in the default services. Thus, when you simulate the robot (`ModelicaStandardLibrary.Modelica.Mechanics.MultiBody.Examples.Systems.RobotR3.FullRobot`) no 3D animation will be shown.

christoff-buerger, 2020-03-05:
Btw, we also have a new Modelica Services for MSL 4.0.0 because some things required changes (I think one is the SI unit handling!). @HansOlsson: I think we should organise how to distribute tool vendor specific `Modelica_Services` properly; this is not only a Dymola issue and we should somehow come up with a proper MAP-LIB solution such that its developers do not run into this stupid problem.

beutlich, 2020-03-05:
There once was a decision to exclude all vendor-specific implementations of ModelicaServices from the MSL.

You could have the ModelicaServices in a separate repo instead.

AHaumer, 2020-03-05:
Thanks @beutlich now it works, but that's really annoying with Modelica_Services:
I you change Modelica_Services and ignore the warning, then open my lib, no conversion is done.
I have to open (and convert) my lib first, then change Modelica_Services.
Maybe that's a tool issue ...
Shall I close this issue?

beutlich, 2020-03-06:
It is a similar tool issue as #3484 for ModelicaExternalC.

tobolar, 2020-03-06:
> Loading the correct ModelicaServices means:

Unfortunately, there is one more step needed. In ModelicaServices 4.0.0, there was added `ModelicaServices.Animation.Vector`. So it may be necessary for some models/examples to also add this empty model to vendor-specific implementation. (I.e. also for those 3.2.3 version from Dymola 2020x.)

beutlich, 2020-03-06:
You can use this Dymola-specific [ModelicaServices](https://github.com/beutlich/ModelicaStandardLibrary/blob/add-Dymola-specific-ModelicaServices/ModelicaServices/package.mo) I needed myself for the creation of the simulation results. I also documented it in the [Wiki](https://github.com/modelica/ModelicaStandardLibrary/wiki/Regression-testing-of-Modelica-and-ModelicaTest#run-the-pedantic-check).

closed 2020-03-06 by beutlich

tobolar, 2020-03-06:
This is a possible but surely not optimal solution.

beutlich, 2020-03-06:
Well, this question comes up with every new release. We simply need to be pragmatic now.

I'll leave it to the new project leaderboard to update the currently detected short-comings w.r.t. processes and the [release cycle](https://github.com/modelica/MAP-LIB_ProjectRules/blob/master/MAP-LIB_ProjectRules.md#annex-7-recommended-release-schedule).

HansOlsson, 2020-03-06:
> Well, this question comes up with every new release. We simply need to be pragmatic now.
> 
We will also try to be better in tools. 
Checking that ModelicaServices.target is correct seems like a good idea; except that it isn't implemented yet.