modelica/ModelicaStandardLibrary #3516: Questionable initialization in 4.0.0 beta 2 reference results for BackSample

maltelenz, 2020-03-19:
In the [reference results](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0-beta.2/Modelica/Clocked/Examples/Elementary/BooleanSignals/BackSample/BackSample.csv#L2) for `Modelica_Synchronous.Examples.Elementary.BooleanSignals.BackSample` the initial value for `shiftSample1.y` is `0`.

In the model [we have](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Clocked/Examples/Elementary/BooleanSignals/BackSample.mo#L17) `backSample1.y_start` set to `true`.

Furthermore, `backSample1.u` is [connected](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Clocked/Examples/Elementary/BooleanSignals/BackSample.mo#L33) to `shiftSample1.y`.

In `Modelica.Clocked.BooleanSignals.Sampler.BackSample` [we have](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Clocked/BooleanSignals/Sampler/BackSample.mo#L12): 
```
Modelica.Blocks.Interfaces.BooleanInput u(start=y_start)
```
Leading to the following (pseudo) equations for the initialization problem:
```
backSample1.y_start = true;
backSample1.u = shiftSample1.y;
backSample1.u.start = backSample1.y_start;
```

The natural solution for that system is:
```
backSample1.y_start = true;
backSample1.u = true
shiftSample1.y = true;
```

So shouldn't `shiftSample1.y` be initialized to `true` (or `1` in the csv world)? This is what we get in SystemModeler.

beutlich, 2020-03-28:
@bernhard-thiele @christoff-buerger @HansOlsson Can you please analyze. Thanks.

HansOlsson, 2020-03-31:
The result reference has another fundamental issue: at some level there is no value for clocked variables like "shiftSample1.y" when the clock doesn't tick. Ignoring that the reference result is correct as far as I understand.

The assumption above is based on solving the initialization problem in the usual way - but https://specification.modelica.org/v3.4/Ch16.html#initialization-of-clocked-partitions state that the normal initialization is not performed, and start-values are only important if previous is used (which isn't the case for `shiftSample1.y`).

This also implies that you can set start-attributes for multiple variables for clocked systems - in 
a way that would have been over-determined if it had been for normal variables. E.g., if we add `Boolean b=previous(shiftSample1.y);` then the start-value for shiftSample1.y matters and we could set it - even though it contradicts the start-value for backSample.

carlj-w, 2020-03-31:
Just a small remark; section 16.9 does not state that start values are only important for variables which has previous applied on them - it says that a variable has _fixed = true_ iff it has previous applied on them. Start values obviously also matters for variables which are used as first argument to _backSample_ (which is the case here.)

One can note that section 16.5.2 does not actually say that that _u_ in _backSample(u, backCounter, resolution)_ will have its start value before the first tick of _u_ (it says that the operator returns the start value of _u_ before the first tick of _u_) - but this seems to be the intention (which means that _backSample.u_ will have a value before its clock has ticked). Nevertheless, the start value will matter (should a variable _u_ used as first argument to _backSample_ then be considered to have _fixed = true_?)

Your point about using previous on _shiftSample.y_ is an interesting one and it makes things clearer. You say that it would be ok to have a system that would be over-determined for normal variables, but in the example you give it wouldn't be overdetermined for normal variables (the start value for _shiftSample.y_, which is fixed, would take priority over the start value for _backSample.u_, which is non-fixed). But, I think I get your point. The equations in the clocked partition do not need to be satisfied by the start values, is this correct?

HansOlsson, 2020-03-31:
> The equations in the clocked partition do not need to be satisfied by the start values, is this correct?

Yes, that is how I interpret it.

beutlich, 2020-04-01:
Can we close this issue as "worksforme"?

christoff-buerger, 2020-04-01:
Sorry for my late reaction. In the ongoing EMPHYSIS project, where we have a lot of clocked models, we are running into similar problems for real practical applications. I hoped I could give at least a glimpse where this may be heading and procrastinated answering this issue for quiet some time hoping for a good answer; unfortunately, this is really a bummer and not thought through in the _specification_.

As @HansOlsson already pointed out, according to the current Modelica language specification, normal initialization is **not** applicable for clocked partitions -- there is only the exception for variables subject to previous, which is more for technical reasons (because we need _some_ value when previous is used the first clock ticks) than defining a sound initialization picture. But I am certain things cannot stay like this with more complex models popping up. For example, for controllers, you like to have a reasonable initialization such that your outputs don't jump after the first clock tick. As it is currently, this is not possible to handle in any reasonable way because there is no proper initialization including outputs -- each hold just states a first start-value such that developers have to guess good start values for control-outputs.

This whole issue is rather a specification shortcoming than a library issue. According to the _current_ Modelica specification, the models are correct.

christoff-buerger, 2020-04-01:
> Can we close this issue as "worksforme"?

We can close as "doesn't work in _practice_ but correct according to the specification". :)

carlj-w, 2020-04-02:
So, I was looking at _Modelica.Clocked.Examples.Elementary.RealSignals.HoldWithDAeffects1_ (and _HoldWithDAeffects1_). Here we have a start value (0.5) for _hold.u_, and the clock starts ticking for _hold.u_ at time = 0.04.

In the references results we don't have a _hold.u_ as a reference signal, but we do have _shiftSample.y_. When the clock ticks, we have _hold.u = shiftSample.y_. According to the reference results, we have _shitSample.y = 0.5_ before the clock ticks for the first time at time = 0.04.

This behavior seems to be the opposite of what we saw in the reference results _Modelica_Synchronous.Examples.Elementary.BooleanSignals.BackSample_ (in _HoldWithDAeffects1__ the start value has been propagated through an equation associated to a clock that has not ticked yet, whereas it didn't in _BackSample_).

In some sense I guess it does not matter what the value of these variables are (or the specification does not care, or the specification considers them to be undefined). If we can agree that this is the case, we should not have these variables as comparison signals.

So, what I am proposing is that variables whose clock does not tick at simulation start and does not have a start value should not be used as comparison signals.

For _HoldWithDAeffects1_ this means that we can use _hold.u_ as a comparison signal (since it has a start value), but not _shiftSample.y_.

beutlich, 2020-04-02:
Can you prepare a PR to remove these signals. Thanks.

HansOlsson, 2020-04-02:
> So, I was looking at _Modelica.Clocked.Examples.Elementary.RealSignals.HoldWithDAeffects1_ (and _HoldWithDAeffects2_). Here we have a start value (0.5) for _hold.u_, and the clock starts ticking for _hold.u_ at time = 0.04.
> 
> In the references results we don't have a _hold.u_ as a reference signal, but we do have _shiftSample.y_. When the clock ticks, we have _hold.u = shiftSample.y_. According to the reference results, we have _shitSample.y = 0.5_ before the clock ticks for the first time at time = 0.04.

I agree with PR for replacing/removing those signals.

To clarify the issue:
In Dymola we don't view clocked signals as existing before (or after) the clock ticks.

However, if you add calls of hold we see that `.hold(shiftSample.y)` is 0 before the first clock tick, whereas `.hold(hold.u)` is 0.5.

You might notice a slight issue above - I will make a separate PR for that.

commit dac0ff7d (2020-04-06): Update Modelica/Resources/Reference/Modelica/Clocked/Examples/Elementary/RealSignals/HoldWithDAeffects2/comparisonSignals.txt

Remove that signal as well. I'm not sure if this completely closes #3516
closed 2020-04-06 by christoff-buerger