modelica/ModelicaStandardLibrary #3518: Questionable reference result for Modelica.Electrical.Analog.Examples.OpAmps.DifferentialAmplifier

maltelenz, 2020-03-20:
In the [reference results](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0-beta.2/Modelica/Electrical/Analog/Examples/OpAmps/DifferentialAmplifier/DifferentialAmplifier.csv#L3769) for the `Modelica.Electrical.Analog.Examples.OpAmps.DifferentialAmplifier`, at time `0.18945` there is a really odd jump in the `opAmp.v_out` variable:

[image]

SystemModeler doesn't reproduce this, and it seems incorrect.

AHaumer, 2020-03-20:
I can't reproduce that odd jump with Dymola 2020 and tolerance 1e-6. 
Maybe the reference result was taken before we changed the tolerance?

beutlich, 2020-03-20:
It is reproducible with tolerance=1e-7.

```
simulateModel(
  problem="Modelica.Electrical.Analog.Examples.OpAmps.DifferentialAmplifier",
  startTime=0,
  stopTime=0.2,
  outputInterval=5e-05,
  method="Dassl",
  tolerance=1e-07,
  resultFile="DifferentialAmplifier");
```

beutlich, 2020-03-21:
I uploaded the new reference file by https://github.com/modelica/MAP-LIB_ReferenceResults/commit/432e32b53c612c5d8448c50516afab67312507b6

Still, I want to keep this issue open as a reminder for @HansOlsson to analyze the solver issue.

[image]

maltelenz, 2020-03-23:
SystemModeler passes correctness against the updated reference results, thank you @beutlich.

beutlich, 2020-03-23:
Thanks for confirmatio. Still a workaround only, since those updated ref results have to created manually.

beutlich, 2020-04-27:
> Still, I want to keep this issue open as a reminder for @HansOlsson to analyze the solver issue.

Closing as tool-issue, finally.

closed 2020-04-27 by beutlich