modelica/ModelicaStandardLibrary #3520: Insufficient reference result quality for Engine1b models

maltelenz, 2020-03-24:
Trying to verify correctness for the model `Modelica.Mechanics.MultiBody.Examples.Loops.Engine1b_analytic` in 4.0.0, I get the following results for `inertia.w`:

[image]

The plot contains the end of the simulation, plotting `inertia.w` for 4 simulations of the same model:

* Simulation run in SystemModeler (WSM) with a high requirement on tolerance and letting DASSL decide the output points (in orange)
* Reference results from the reference repository (which are sampled with 5000 points, in blue)
* Simulation run with WSM default settings for tolerance and output points (2000 points, in green)
* Simulation run with OpenModelica with their default settings (500 points, in red)

As can be seen, there are some serious problems:

* The reference results are not tightly enough sampled to cleanly represent the sinusoidal signal
* There is a time offset in the signal when run in different tools and with different solver settings (the low number of samples makes this hard to see though)
* The default settings in the tools where I'm aware of what they are are not enough to even visually represent the signal

Can Dymola pass correctness against its own baseline when run with default settings?
If not, I think the experiment setting for the model should be adjusted until it can.

The model `Modelica.Mechanics.MultiBody.Examples.Loops.Engine1b` shows similar behavior.

sjoelund, 2020-03-24:
I believe @casella is collecting the regressions for OpenModelica https://libraries.openmodelica.org/branches/newInst/Modelica_trunk/files/Modelica_trunk_Modelica.Mechanics.MultiBody.Examples.Loops.Engine1b.diff.inertia.w.html fails with the 4.0.0 reference files, but it used to work with the 3.2.3 reference files and the model from the master branch.

maltelenz, 2020-03-24:
@sjoelund Out of curiosity, what does "high" and "low" mean in the linked plot?

sjoelund, 2020-03-24:
If the signal is within high and low, we consider it to be the same as the reference. In the case of this engine, the problem is that the period in this signal becomes shorter and shorter, but the reference files does not have tighter samples here. Having more samples should solve the problem.

beutlich, 2020-03-28:
@christoff-buerger In e6c7cca59cd829f67b4d3c575b9a663ab2b0a0c0 you increased the simulation time of the Engine1b models from 0.5 to 5 seconds (to match the simulation time of the Engine1a model) which no introduces the high-frequency oscillations. I propose to reduce the simulation time again. See #3527.

PR #3527 (2020-03-30): Decrease simulation time of Engine1b models
Closes #3520
closed 2020-03-30 by beutlich