modelica/ModelicaStandardLibrary #3521: Superfluous file in reference results

maltelenz, 2020-03-24:
Is there a reason for the presence of this file, or is it just an accident?

https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0-beta.2%2B412d17023/ModelicaTest/Media/TestOnly/R134a_setState_pTX/MoistAir.csv

beutlich, 2020-03-24:
Can you please create a PR to remove that (and other) CSV file(s). Thanks.

maltelenz, 2020-03-24:
I guess I should push my pull request branch directly to the modelica/MAP-LIB_ReferenceResults repo, since it is a private repository?

beutlich, 2020-03-24:
Ah, wrong repo. Let me see if I can transfer the issue.

beutlich, 2020-03-24:
Done directly in browser: https://github.com/modelica/MAP-LIB_ReferenceResults/commit/5b013acbeef53b432faccbdac8d7dbabe5c104b4

closed 2020-03-24 by beutlich

maltelenz, 2020-03-24:
For what it's worth, this was the only weird csv file I found.

beutlich, 2020-03-24:
Probably some leftover when I tested locally. And CSV is not in gitignore for reasons 😄