modelica/ModelicaStandardLibrary #3523: Jumps in signal at very tiny time intervals in PolyphaseTwoLevel_RL reference results

maltelenz, 2020-03-24:
In the [reference results](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0-beta.2/Modelica/Electrical/PowerConverters/Examples/DCAC/PolyphaseTwoLevel/PolyphaseTwoLevel_RL/PolyphaseTwoLevel_RL.csv) for `Modelica.Electrical.PowerConverters.Examples.DCAC.PolyphaseTwoLevel.PolyphaseTwoLevel_RL` there are two subsequent jumps at an extremely short distance from each other. You can see this in rows 3433-3436 in the linked csv file (can't direct link because github restrictions on file size).

The variables in question are `voltageSensor.v[1]` through `voltageSensor.v[3]`. Plotted:
[image]
Please ignore the labeling on the x axis (bug in SystemModeler).

As you can see, SystemModeler (blue) only has a single event iteration leading to the complete jump, while the reference result (orange) takes two jump very close to each other.
The reference reports jumps at `0.030250000000000003` and `0.030250000000115126`, a distance of approximately 1.15e-13 apart.

Is there anything in the model that explains the two separate jumps? Is it expected? Do other tools reproduce these two jumps?

AHaumer, 2020-03-25:
I can't reproduce that with Dymola, I can't zoom in beyond a resolution of ~1e-7.
I can't say that this is expected, but IMHO it's possible - depending on tolerance and event handling. 
There are time events (fundemantalWaveVoltage, fundemantalWaveCurrent, signalPWM) and state events (signalPWM, inverter). It could happen that a state event is iterated within a very short time span before or after a time event and both results are stored. Does the behaviour change with another choice of the solver and / or the tolerance?
Is the difference marked by the comparison tool? 
IMHO it's neglectible. Sorry I don't have time to dog deeper - we are 150% busy organizing online content for our students.

beutlich, 2020-04-27:
Clsoing as worksforme, finally.

henrikt-ma, 2021-01-14:
Instead of closing as _worksforme_, shouldn't the correct action be to remove the `voltageSensors.v` signals from the comparison variables?  It would comply with @AHaumer's suggestion to not dig deeper, and would still leave us with 44 less problematic variables to be compared.

TManikantan, 2023-03-09:
@christiankral Hi ,I recently assumed the role of Project Deputy, would please take time look at it? Do contact me for any help that i could provide with this ticket.

TManikantan, 2023-05-15:
@christiankral @AHaumer what is your suggestion for this issue?Do you agree with https://github.com/modelica/ModelicaStandardLibrary/issues/3523#issuecomment-760084305

TManikantan, 2023-07-06:
As per suggestion from @henrikt-ma in the above comment,i have to remove `voltageSensors.v` from [Modelica/Resources/Reference/Modelica/Electrical/PowerConverters/Examples/DCAC/PolyphaseTwoLevel/PolyphaseTwoLevel_RL/comparisonSignals.txt](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Resources/Reference/Modelica/Electrical/PowerConverters/Examples/DCAC/PolyphaseTwoLevel/PolyphaseTwoLevel_RL/comparisonSignals.txt). I am confused should the [result_csv](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0/Modelica/Electrical/PowerConverters/Examples/DCAC/PolyphaseTwoLevel/PolyphaseTwoLevel_RL/PolyphaseTwoLevel_RL.csv) also be changed? if yes, is it fine to just delete the columns from the  [result_csv](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0/Modelica/Electrical/PowerConverters/Examples/DCAC/PolyphaseTwoLevel/PolyphaseTwoLevel_RL/PolyphaseTwoLevel_RL.csv) or to generate a new set of result.
@henrikt-ma ,  @christiankral your suggestions would be really helpful.

beutlich, 2023-07-06:
> I am confused should the [...]

No, the results will be regenerated later when the new release is ready, usually for beta and rc releases.

TManikantan, 2023-07-06:
> No, the results will be regenerated later when the new release is ready, usually for beta and rc releases.

thank you for your response.

casella, 2024-01-16:
The discussion continued on the linked PR #4167.

closed 2024-01-19 by arunkumar-narasimhan