modelica/ModelicaStandardLibrary #3530: Clock variables in reference results

maltelenz, 2020-04-01:
There are cases where we have clock variables (note, not clock**ed** variables) in reference results. [One such example](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Resources/Reference/Modelica/Clocked/Examples/Elementary/ClockSignals/SubSample/comparisonSignals.txt#L3) is `subSample.y` in `Modelica.Clocked.Examples.Elementary.ClockSignals.SubSample`.

In the reference file, this is represented as a constant zero.

In SystemModeler, we have chosen to represent ticks in clock variables as an increasing integer, increasing by 1 each time the clock ticks. This seems more useful for the user, as opposed to a constant zero. We have not found anything clear and definitive in the spec that forces the choice of the value of clock variables.

I see two choices:

1. We agree that representing a clock variable with increasing value as it ticks is useful, and we update the reference data.
2. We can't agree on the representation, or we agree that the constant value is correct, meaning having the variable as a reference signal serves no purpose, and we remove the variable from comparison.

I prefer 1, since it adds value, but if we can't agree on that we should do 2.

HansOlsson, 2020-04-01:
I would prefer to remove it - as I don't think we can agree on a representation now.
And technically clock variables aren't a good fit for our current comparisons, as they are only defined when the clock ticks - which means that they don't really exist for most of the trajectory. For clock**ed** variables we can circumvent that by saying that we don't store x but hold(x), but there is nothing similar for clock variables.

Note that Dymola does indirectly store the increasing value - but internally linked from all variables that are on that (sub-)clock.

maltelenz, 2020-04-01:
> I would prefer to remove it - as I don't think we can agree on a representation now.

I suspected as much.

I should be in a good place to prepare a pull request tomorrow to remove all clock variables from the lists of comparison signals, if that is what we wish to do.

beutlich, 2020-04-01:
I was not aware of this issue when I created the comparisonSignals.txt for Modelica.Clocked as part of https://github.com/modelica/ModelicaStandardLibrary/pull/3348#issuecomment-584614256 w/o feedback from the library officers.

> I should be in a good place to prepare a pull request tomorrow to remove all clock variables from the lists of comparison signals, if that is what we wish to do.

Yes, please go ahead.

PR #3531 (2020-04-02): Remove clock signals from comparison signals as agreed in #3530.
Fixes #3530
closed 2020-04-02 by beutlich