modelica/ModelicaStandardLibrary #3535: Some tag names are not semantic versions

sjoelund, 2020-04-06:
We have for example tags:

```
v3.2.3+build.3-beta.1
v3.2.3+build.3-beta.2
```

But according to https://semver.org/spec/v2.0.0.html, they should have been:
```
v3.2.3-beta.1+build.3
v3.2.3-beta.2+build.3
```
Is there a good reason for this?

beutlich, 2020-04-06:
> Is there a good reason for this?

No. Apologies for my ignorance. Shall we rename these tags?

beutlich, 2020-04-06:
Hm, according to https://regex101.com/r/vkijKf/1/ and the mentioned RegEx they should be valid, too.

sjoelund, 2020-04-06:
https://semver.org/#is-there-a-suggested-regular-expression-regex-to-check-a-semver-string says they shouldn't be used. This might have been what changed in semver 2.0.0 though...
I think it's probably good to at least add the correct tags; unsure if the other ones should remain or not...

beutlich, 2020-04-06:
The second RegEx of https://semver.org/#is-there-a-suggested-regular-expression-regex-to-check-a-semver-string is the one that works for the current tags.

sjoelund, 2020-04-06:
Hmmm... It seems to be a valid identifier, but the meaning is then not a prerelease. It's a single build information that happens to contain a hyphen...

dietmarw, 2020-04-06:
I agree with @sjoelund  and it slipped my attention too. After all we are doing a new release of `v3.2.3+build.3` of which some beta versions are tagged.

sjoelund, 2020-04-07:
Thinking about it, I guess since the old, bad tags are wrong but no invalid and will sort higher than the actual release... they should probably be removed.

beutlich, 2020-04-07:
I read `v3.2.3+build.3-beta.1` from right to left as first beta of the third build of the 3.2.3 release. That is also intended interpretation. For `v3.2.3-beta.1+build.3` it is harder to get that interpretation.

sjoelund, 2020-04-07:
That's because builds aren't supposed to be new versions. But according to semver, `v3.2.3+build.3-beta.1` is a build of the release called `build.3-beta.1`. And `v3.2.3-beta.1+build.3` is the third build of `beta.1`; it also won't sort the way you want but will at least be treated as a pre-release.

beutlich, 2020-04-07:
I edited my above post.

sjoelund, 2020-04-07:
I think the problem is once you start saying the version number is according to semver (with 4.0.0), and people (like me) try to fetch all releases and figure out which are the latest stable releases.

beutlich, 2020-04-07:
So what to do? Leave the invalid tags as now or update the tags and get the undesired sorting?

sjoelund, 2020-04-07:
As far as I can tell, you have the undesired sorting anyway, because
`v3.2.3+build.0-beta.1` > `v3.2.3+build.0` (due to the rule that identifiers with only numbers are always sorted as lower than other identifiers):
> Identifiers consisting of only digits are compared numerically and identifiers with letters or hyphens are compared lexically in ASCII sort order. Numeric identifiers always have lower precedence than non-numeric identifiers.

henrikt-ma, 2020-04-07:
Since semver.org is only mentioned as source of inspiration, shouldn't we be explicit and write out the form of the version numbers used by MSL, even if it happens to be the very same form used by semver.org?

sjoelund, 2020-04-07:
Actually, I'm wrong again. The build information is ignored when comparing versions... So they would all sort to the same thing. Maybe it would be OK to say how to sort based on the build information in that case... Maybe.

dietmarw, 2020-04-07:
We really starting to **use** semver (inspired) from v4.0.0 on. Everything before was kind of a hack since there build is actually a PATCH level release. But we tried to have a orderly sequence of version numbers anyway. So this is just something we have to live with.

beutlich, 2020-04-07:
So closing as worksforme?

dietmarw, 2020-04-07:
Yes since the `v3.2.3+build.3-beta.1` scheme is the one we adapted prior real semver compatibility but this scheme was applied consistently.

closed 2020-04-07 by dietmarw

beutlich, 2020-04-07:
> 
> 
> Since semver.org is only mentioned as source of inspiration, shouldn't we be explicit and write out the form of the version numbers used by MSL, even if it happens to be the very same form used by semver.org?

We already did so, see

https://github.com/modelica/ModelicaStandardLibrary/blob/7b0253071a017fc08ad323676a27f3c1f2dc2bc0/Modelica/package.mo#L2244-L2245

and

https://github.com/modelica/ModelicaStandardLibrary/blob/7b0253071a017fc08ad323676a27f3c1f2dc2bc0/Modelica/package.mo#L2365-L2367

henrikt-ma, 2020-04-07:
None of those places define the use of pre-release versioning and build metadata, which is the whole problem of this issue.

sjoelund, 2020-04-07:
Yes, I think the problem is nowhere is the `3.2.x` version of it described. For 4.0.0, it should be fine since the builds are not new versions.

henrikt-ma, 2020-04-07:
By the way, in view of the long planned standardization of semantic versioning of Modelica libraries, this wouldn't have been considered an issue since library versions released before semantic versioning was introduced in the language wouldn't be treated semantically anyway. (At least not according to the Modelica specification; tools might still determine it helpful to users to guess version semantics based on the style of the version string.)

I think this case is similar; all Modelica tools should now about the different versioning schemes used by MSL through the history, and add the necessary intelligence in the tool to determine how the various versions relate to one another.  Lacking standardization in the Modelica language, this task will be easier the more clearly the MSL defines its use of semantic versioning.