modelica/ModelicaStandardLibrary #3536: Very long filenames in Modelica/Resources

maltelenz, 2020-04-07:
Long filenames and paths [are a problem on Windows](https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file#maximum-path-length-limitation), with a practical limit in the region of 260 characters.

We have some very long path names in the library, which results in real problems when trying to work with, and distribute, the library.

Ignoring the new `comparisonSignals.txt`, which can be skipped when shipping the library to end users, all the longest file paths are in `Modelica/Resources/Images/Fluid/Dissipation/Utilities/\
SharedDocumentation/`.

The longest path in 3.2.3 was 130 characters long.
Below are all the paths that are longer than that.

Could we move these files to something shorter, and have policies that keep the file path length short(er)?
These files will be inside some folder which also wants part of the path length.
This is basically the minimum length for a regular installation on Windows for a tool shipping the library:
`"C:/Program Files (x86)/Wolfram SystemModeler 12.2/Libraries/Modelica 4.0.0/"`
which is already 75 characters. Add the 153 from the library, and we are already using 228 of our allocated 260 characters, leaving very little room for decisions where to put the library files in a tool installation.

The list of files with longer names than in 3.2.3, and their length:
```
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/General/fig_general_dp_nominalPressureLossLawDensity_MFLOWvsDP.png->153
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/General/fig_general_dp_pressureLossCoefficient_MFLOWvsDP.png->147
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/General/fig_general_dp_nominalDensityViscosity_MFLOWvsDP.png->147
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/General/fig_general_dp_nominalDensityViscosity_DPvsMFLOW.png->147
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Valve/fig_valve_dp_severalGeometryOverall_zetaVsOpening.png->146
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/StraightPipe/fig_straightPipe_dp_overall_lambdavsRe_ver.png->146
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Channel/fig_channel_dp_internalFlowOverall_lambdaVsRe.png->144
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/StraightPipe/fig_straightPipe_laminar_lambdavsRe_ver.png->143
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Valve/fig_valve_dp_severalGeometryOverall_MFLOWvsDP.png->142
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/StraightPipe/fig_straightPipe_dp_overall_MFLOWvsDP.png->141
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/StraightPipe/fig_straightPipe_dp_overall_DPvsMFLOW.png->141
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/StraightPipe/fig_straightPipe_dp_laminar_DPvsMFLOW.png->141
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Orifice/fig_orifice_thickEdgedOverall_ZETAvsLENGTH.png->141
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Bend/fig_bend_dp_curvedOverall_DPvsMFLOWwrtDelta.png->139
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Orifice/fig_orifice_thickEdgedOverall_MFLOWvsDP.png->138
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Orifice/fig_orifice_thickEdgedOverall_DPvsMFLOW.png->138
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/General/fig_general_dp_volumeFlowRate_MFLOWvsDP.png->138
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Bend/fig_bend_dp_edgedOverall_MFLOWvsDPwrtDelta.png->138
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Bend/fig_bend_dp_edgedOverall_DPvsMFLOWwrtDelta.png->138
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Bend/fig_bend_dp_curvedOverall_DPvsMFLOWwrtRD.png->136
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Orifice/fig_orifice_suddenChangeContraction.png->134
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/StraightPipe/fig_straightPipe_kc_turbulent.png->133
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Orifice/fig_orifice_suddenChangeExpansion.png->132
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/General/fig_general_dp_idealGas_MFLOWvsDP.png->132
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/PressureLoss/Bend/fig_bend_dp_curvedOverall_DPvsMFLOW.png->131
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/StraightPipe/fig_straightPipe_kc_overall.png->131
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/StraightPipe/fig_straightPipe_kc_laminar.png->131
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/StraightPipe/fig_kc_twoPhaseOverall_KC_4.png->131
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/StraightPipe/fig_kc_twoPhaseOverall_KC_2.png->131
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/HelicalPipe/fig_helicalPipe_kc_turbulent.png->131
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/General/fig_kc_approxForcedConvection_T3.png->131
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/General/fig_kc_approxForcedConvection_T2.png->131
Modelica/Resources/Images/Fluid/Dissipation/Utilities/SharedDocumentation/HeatTransfer/General/fig_kc_approxForcedConvection_T1.png->131
```

maltelenz, 2020-04-07:
After having dug around in history, this is a request to change the decision in #2716.

dietmarw, 2020-04-07:
As Linux user I could not care less. Sorry ;-)

HansOlsson, 2020-04-07:
> After having dug around in history, this is a request to change the decision in #2716.

I don't see how the decision #2716 can be the direct cause of this. On the contrary I would say that we should better implement the decision in #2716.

That ticket was an attempt of loosely interpreting "The PNG files should be placed in a folder which exactly represents the package structure." for #2695.

What I proposed in #2716 was to not have directories with just one file - but the total path length is unchanged by that (or possibly even shortened).

As I see it the issue is that the file-names for the FluidDissipation in themselves are excessively long e.g, `fig_general_dp_nominalPressureLossLawDensity_MFLOWvsDP.png`, and the "fig_general" is a duplicate going against the idea in #2716, and as we only have one image in a model we don't even need the extra "_MFLOWvDP.png" to disambiguate it.

The class name also seems excessive long.

maltelenz, 2020-04-08:
> As Linux user I could not care less. Sorry ;-)

I mean, me too, but unfortunately we have customers who haven't realized what a broken heap of legacy Windows is :)

> I don't see how the decision #2716 can be the direct cause of this. On the contrary I would say that we should better implement the decision in #2716.

The mirroring of package structure already results in 110 characters for the longest case:
`Modelica/Resources/Images/Magnetic/FluxTubes/Examples/Hysteresis/Components/Transformer3PhaseYyWithHysteresis/`. Some leeway here, maybe not requiring all the pieces of the class path, maybe would be a good idea. Granted, it would only save us 13 characters, since the actual Modelica package in `Modelica/Magnetic/FundamentalWave/Examples/BasicMachines/SynchronousMachines/ComparisonPolyphase/` reaches 97 characters (plus actual .mo filename lengths).

> As I see it the issue is that the file-names for the FluidDissipation in themselves are excessively long

Upon further inspection, I agree that seems to be the main issue. The pull request nicely demonstrates this, as it brings down the max path length to 131, just a single character more than 3.2.3. I think we can live with that.

PR #3537 (2020-04-08): Shorten image names of Modelica.Fluid.Dissipation
I also edited/fixed some white spaces. Best is to watch the diff ignoring the white space changes.

Resolves #3536.
closed 2020-04-08 by beutlich