modelica/ModelicaStandardLibrary #3539: Invalid connection line points in DirectCapacity

maltelenz, 2020-04-08:
There are some connection lines with three elements in a point in `Modelica.Thermal.HeatTransfer.Examples.Utilities.DirectCapacity`. I assume it's a typo?

https://github.com/modelica/ModelicaStandardLibrary/blob/7b0253071a017fc08ad323676a27f3c1f2dc2bc0/Modelica/Thermal/HeatTransfer/Examples/Utilities/DirectCapacity.mo#L25-L30

HansOlsson, 2020-04-08:
The line below the marked ones also have the same issue.

beutlich, 2020-04-08:
@HansOlsson Could you please create a PR for it. Thank you.

HansOlsson, 2020-04-08:
I'm looking into it. 

So far I found that it was introduced by:
https://github.com/modelica/ModelicaStandardLibrary/commit/07da25fe6ba7c30539e4ec4c60a8336e071bffad
so I now know what the intent was.

beutlich, 2020-04-08:
Ah, now I get it.

PR #3541 (2020-04-08): Correct illegal annotations.
Closes #3539
I checked all of MSL for illegal annotations that's why I got one more. Note: the check is not yet complete so I cannot guarantee that there are no additional issues.

In Valves I believe the Placement is leftover from some component, possibly opening_min that was transformed into a parameter, and thus no longer needs a Placement-annotation.
closed 2020-04-08 by beutlich