modelica/ModelicaStandardLibrary #3543: Incorrect initialization of SinglePhaseTransformerWithHysteresis1

henrikt-ma, 2020-04-22:
The initialization of `Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis1` doesn't seem to agree with the [reference result](https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0-beta.2%2B412d17023/Modelica/Magnetic/FluxTubes/Examples/Hysteresis/SinglePhaseTransformerWithHysteresis1/SinglePhaseTransformerWithHysteresis1.csv).

In the reference result, `core.B` starts at 0.57853872532305728, but SystemModeler thinks that it should be 0.0, derived in two simple steps:
- `core.Phi` is initialized to its `start` value, which is 0.0.
- `core.B` is then initialized from the equation `core.Phi = core.B * core.A`, where `core.A` is a non-zero parameter.  The solution to this is `core.B` = 0.0.

beutlich, 2020-04-25:
There is an initial equation for B

https://github.com/modelica/ModelicaStandardLibrary/blob/cbc5092c9b0cf75d5a714028ad30c4c140fdfc18/Modelica/Magnetic/FluxTubes/BaseClasses/GenericHysteresisTellinen.mo#L19-L21

which initializes as follows

```
(core.B|0.57853872532305728):=0.5*(core.diffHyst|2.3140331597371091)*(core.MagRel|0.5)+0.5*(core.hystF|1.1570470152573344)+0.5*(core.hystR|-1.1569861444797744)
```

henrikt-ma, 2020-04-27:
You are right.  Today, I have no idea where I got that reported structure of the initialization, and SystemModeler is mysteriously initializing the simulation in agreement with the reference result.  Closing as _Invalid_.

closed 2020-04-27 by beutlich

henrikt-ma, 2020-04-27:
For the record, the problem I reported here was (and it doesn't come as a surprise) due to #3409. I mixed up two seemingly similar versions of the same test that, but where the fix for #3409 isn't present in one of them.