modelica/ModelicaStandardLibrary #3544: strtol_l only present if __USE_GNU defined

d94pn, 2020-04-28:
In ModelicaIO.c the following logic decides which strtol-function to use:

if !defined(NO_LOCALE) && (defined(_MSC_VER) && _MSC_VER >= 1400)
     nRow = (unsigned long)_strtol_l(token, &endptr, 10, loc);
#elif !defined(NO_LOCALE) && (defined(__GLIBC__) && defined(__GLIBC_MINOR__) && ((__GLIBC__ << 16) + __GLIBC_MINOR__ >= (2 << 16) + 3))
        nRow = (unsigned long)strtol_l(token, &endptr, 10, loc);
#else
        nRow = (unsigned long)strtol(token, &endptr, 10);
#endif

However, for at least gcc 5.3 on OpenSuse (probably other targets as well), this results in the following kind of warnings:

ModelicaIO.c:784:31: warning: implicit declaration of function ‘strtol_l’ [-Wimplicit-function-declaration]
         nRow = (unsigned long)strtol_l(token, &endptr, 10, loc);

The reason is that in stdlib.h, strtol_l are not declared unless __USE_GNU is defined.
Hence, the condition for using strtol_l should involve "defined(__USE_GNU)".

Then again, during linking strtol_l is actually found, so one could argue that the price of the ugly warning is worth to get the strtol_l function.

I have no solution to this but there is an inconsistency to handle here. Note also that the same applies to other locale functions such as strtod_l.
It also applies to other C functions such as ModelicaStrings.c.

beutlich, 2020-04-28:
Hm, `_GNU_SOURCE` is defined at the very beginning of the file, i.e., before including stdlib.h.

https://github.com/modelica/ModelicaStandardLibrary/blob/cbc5092c9b0cf75d5a714028ad30c4c140fdfc18/Modelica/Resources/C-Sources/ModelicaIO.c#L78-L80

@HansOlsson told me some time ago, that you have special build process, where all C files are included in one meta C file. Could that be again the reason for the linker trouble?

d94pn, 2020-04-29:
But I cannot find any connection from _GNU_SOURCE to __USE_GNU; the latter remains unset when entering stdlib.h.
Btw, it is not a linker trouble as the linking part succeeds.

beutlich, 2020-04-29:
Isn't there in your featues.h?

```c
#ifdef	_GNU_SOURCE
# define __USE_GNU	1
#endif
```

d94pn, 2020-04-29:
Aha, now I see the problem: Our main compilation units includes ModelicaIO.c, but have already included features.h prior to that, where _GNU_SOURCE is not set. Conclusion: The problem is one our side. Thanks for help in sorting this out.

closed 2020-04-29 by d94pn

beutlich, 2020-04-29:
Indeed, including all C files in one single compilation unit was causing the trouble again. And actually, it is not the proposed build process as you can see from the provided build projects where we provide make files etc.