modelica/ModelicaStandardLibrary #3545: Use unbounded=true attribute in mechanical interfaces?

mestinso, 2020-04-29:
It seems to me that the new (Modelica 3.4) unbounded attribute is applicable for mechanical interfaces. For example, in Modelica.Mechanics.Rotational.Interfaces.Flange, I would think unbounded=true should be added to phi:
```
SI.Angle phi(unbounded=true) "Absolute rotation angle of flange";
```
Alternatively, it could be added to the Angle type itself, but that may be too presumptive.

Thoughts?

beutlich, 2020-04-29:
Definitely. I also raised the topic in #2946 and even mentioned the Real.unbounded attribute at the [kick-off meeting](https://svn.modelica.org/projects/ModelicaDesign/trunk/MA-Projects/MAP-LIB/MeetingMinutes/20190320-MSL-v4.0.0-Kick-Off/190320_MSL_Kick-Off.pdf) 13 months ago. But, library officers of mechanical libraries (@MartinOtter or @tobolar) did not reply so far.

MartinOtter, 2020-05-04:
Please, do not introduce unbounded=true to MSL:

- There are two application areas: In one area the angle is controlled to follow a reference angle (e.g. robots, elevators, and in general positional drives) where unbounded must be false, and in other cases the shaft is permanently rotating and the angle is of no importance at all (e.g. drive train of a vehicle) and unbounded should be true here. Since neither of the areas is more important / common as the other one, the default value of unbounded (=false) should be used.

- unbounded=false does not give a wrong solution, but it might be unefficient (so the simulation is slower as it could be). unbounded=true can give wrong simulation results, if error control is switched off for variables that are not unbounded but have to follow a range-limited reference signal. Therefore unbounded=false is the much better default. 

- There are many existing applications of MSL. Suddenly, switching off the error control of variables by having a default of unbounded=true, may give different or even wrong simulation results of existing models.

mestinso, 2020-05-06:
@MartinOtter: while I understand your point that the two different application areas mentioned have different needs and priorities, I don't understand why you say unbounded "must be false" for the case of positional drives? As far as I can tell, `unbounded=true` is more conservative since `tol*(abs(nominal)+(if x.unbounded then 0 else abs(x)))` is generally a tighter tolerance when `x.unbounded=true`. Furthermore, it leads to a consistent tolerance across the positional space, which seems desirable in the case of positional drives.

As I understand,`unbounded=true` simply removes relative error control while maintaining absolute error control (as indicated in the above equation). Therefore, to me, we just need to ask ourselves if relative error control makes sense or not for the given variable. One way to make this determination is to see if it has the potential to grow without bound (thus the name), but another way is to check if it has a meaningful, non-arbitrary zero value. Another thought experiment is you can ask yourself if the absolute tolerance should grow as the value moves away from the thing we call zero. For items like pressure, temperature, mass, current, power, heat flow rate, etc., these all have a meaningful zero, and it is desirable/acceptable for the absolute tolerance to increase as their absolute values get large.

However, in the case of things like position, rotational angle, or even time (conceptually), the thing we call zero is arbitrary, so I wouldn't want to increase my absolute tolerance as I move away from some arbitrary origin. Therefore I think they should have the unbounded=true attribute.

Wikipedia seems to have some relevant discussion on this:
https://en.wikipedia.org/wiki/Approximation_error#Uses_of_relative_error
https://en.wikipedia.org/wiki/Level_of_measurement#Ratio_scale
Using the terminology in those sources, my claim is that relative error control makes sense for variables measured on a "ratio scale", but not for variables measured on an "interval scale".

MartinOtter, 2020-06-07:
"unbounded" is a fact that is stated for a variable v. If unbounded=false, v will stay within some bounds, say vmin <= v <= vmax. For a technical device, nearly all variables are bounded, since the device is designed for an operating region and the operating region is typically bounded. For example, the angle of a robot joint is bounded, e.g. -180 deg <= angle <= 180 deg. There are only some rare cases where a variable of a technical device is not bounded, for example, the absolute position of a vehicle or the angle of an "endless axis" or time. 

How the actual value of "unbounded" is utilized, is a tool issue. 

For a simulation program, it would be natural to utilize "unbounded" for the step-size control. A typical intergrator uses relative and absolute tolerances for the step-size control. This corresponds to unbounded=false. This means, if unbounded=true is set for a state variable, the step-size control will most likely change (depending on the tool), giving slightly different results or changing performance. In order that the simulation behaviour of exisiting models remains the same with a new MSL version, unbounded=false must be the default in MSL. Only for new models introduced in a new MSL version, unbounded=true can be used.

Integrators typically estimate the local error `err_i` of a state `x_i` and adapt the step-size so that the local error is bounded:

```
|err_i|  < tolrel*|x_i| + tolabs_i
```

In Modelica tools, `tolabs_i` is computed from tolrel and from the nominal value, e.g., 

```
tolabs_i = fac*tolrel*x_i.nominal   // fac = 1.0 or 0.1
```

and therefore

```
|err_i|  < tolrel*|x_i| + fac*tolrel*x_i.nominal
```

If `unbounded=true`, a relative error control makes usually no longer sense and therefore the relative error control `tolrel*|x_i|` should be switched off leading to:

```
|err_i|  < (if unbounded then 0 else tolrel*|x_i|) + fac*tolrel*x_i.nominal
```

> As far as I can tell, unbounded=true is more conservative 

Yes, but by purpose all integrators utilize relative tolerances to speed-up simulation. If only absolute tolerances would be used, the result could be much too precise and therefore the simulation time much too long.