modelica/ModelicaStandardLibrary #3556: MinGW possibly neglected in new safe.math.h

d94pn, 2020-05-20:
Hi,
When testing the source code in MSL 4.0.0, rc 2, I got compiler errors on MinGW:

safe-math.h:100:23: fatal error: intsafe.h: No such file or directory

Perhaps the MinGW platform has not yet been considered in the preprocessing?

beutlich, 2020-05-20:
Strange. I build the binaries for the gcctdm510 directories and also build matio using MinGW on [Appveyor](https://ci.appveyor.com/project/tbeu/matio/branch/master) without issues. I guess I need some more information on your build environment than just the plain error.

d94pn, 2020-05-20:
I actually tested with a quite old MinGW version. Findings:

1. gcc.exe (GCC) 4.8.1, MINGW32_NT-6.2 fails.
2. gcc.exe (MinGW.org GCC-6.3.0-1) 6.3.0 succeeds.

So the error is probably less critical. Also, I do not know how to check the version of MinGW gcc via preprocessor variable. If possible, the first of these lines in ModelicaMatIO.c:

```
#if !defined(__BORLANDC__) && !defined(__LCC__) && !(defined(_MSC_VER) && _MSC_VER < 1400) /* && !defined(__MINGW32__)*/
#define HAVE_SAFE_MATH 1
```

is a candiate for such checking I assume.

beutlich, 2020-05-20:
> Also, I do not know how to check the version of MinGW gcc via preprocessor variable.

It is an GCC, so we can use `__GNUC__` (and variants) along with `__MINGW32__`.

beutlich, 2020-05-20:
https://github.com/modelica/ModelicaStandardLibrary/blob/9821b66f09364c62c13fc349b5df5fb875bef148/Modelica/Resources/C-Sources/safe-math.h#L22-L28

The issue probably is, that old GCC has no `__has_include` why it then sets `PSNIP_SAFE_HAVE_INTSAFE_H` in the else branch.

Did it help to set `PSNIP_SAFE_FORCE_PORTABLE`?

d94pn, 2020-05-20:
I can confirm that setting `PSNIP_SAFE_FORCE_PORTABLE` helped gcc 4.8.1.
Also, for gcc 5.3.0 it works without modifications. In this case, the macro `__has_include` was indeed set, but it was not for gcc 4.8.1.

PR #3568 (2020-06-04): Do not include intsafe.h on MinGW
@d94pn Can you please confirm with (GCC) 4.8.1, MINGW32_NT-6.2 and `PSNIP_SAFE_FORCE_PORTABLE` not defined.

closes #3556
PR #3572 (2020-06-05): Do not include intsafe.h on MinGW
Back-port #3568 to fix #3556 in maint/4.0.1.
closed 2020-06-04 by beutlich