modelica/ModelicaStandardLibrary #3558: Error: "Corrupt file. nvar 0" with simulation

trilbytim, 2020-05-20:
I'm getting a somewhat intermittent problem, but I've tried to recreate with a simple model here, a tube bouncing on two spring/dampers. When I simulate this with the default settings of start time 0, stop time 1 and Number of Intervals 500 it works, I have data I can plot and the visualisation works. However when I increase the Stop time to 10 and the Number of Intervals to 2000 it (usually, but not always!!!) fails to produce data. The simulation appears to run, it reaches 100% and there are no error messages. However when I try to plot anything I get a pop up error "OMEdit - Error: Corrupt file. nvar 0" Even if I change the simulation settings back to Stop time 1 and Number of Intervals 500 and re run the simulation I still don't have any data to plot.
If I try to do the "Simulate with Animation" option then OMEdit crashes completely and shuts down.
This behaviour seems erratic and connected to the Number of Intervals. Sometimes 1000 intervals will create the error, sometimes it doesn't. 2000 intervals almost always seems to create, but sometimes even that has worked.
[scratch.zip](https://github.com/modelica/ModelicaStandardLibrary/files/4657750/scratch.zip)

goatchurchprime, 2020-05-20:
That crash is not reproducing for me now no matter how many times I try.  The more complex frame_model we started with is also not crashing.  You can make it "Save experiment annotation inside model" on the Simulation setup to set the Stop time and Number of Intervals.

dietmarw, 2020-05-20:
This looks like a tool issue so you should report it at https://trac.openmodelica.org/OpenModelica/

closed 2020-05-20 by dietmarw

goatchurchprime, 2020-05-21:
Oh man. It's on trac.  Really?