modelica/ModelicaStandardLibrary #3566: Is UTHASH overkill?

DagBruck, 2020-05-27:
Sorry to re-hash (pun intended) an old bit of code...

MSL current uses `uthash.h` for two reasons, if I understand correctly:

1. To provide a hash function for Modelica strings.
2. To implement shared tables, so if you reference the same table several times you just read it once.

The problem, in my view, is that UTHASH is really a terrible piece of code, it consists of 1238 lines of C macros. So it is a big chunk of code, there are of course the risk of the macros being bungled or causing conflicts, and it is a nightmare if we think we need to support it.

Also, I think that maybe it is overkill for the things we use it for. To hash strings we can take 5 lines of code to get a very reasonable hash function. For the tables, well the number of tables in a model is rather limited and it is in any case only used when you read the table (not during table lookup).

Proposal: simplification, especially for Modelica strings.

beutlich, 2020-05-27:
Uthash actually also is used in ModelicaInternal.c to cache the opened files. Thus, it is utilized in three places. It was first added about 6 years ago by 01a172d6e29b762b897a29658ccdfd226a05135f for #1550 and released later by MSL v3.2.2.

I actually do like uthash.h as it is. It is well maintained, tested, documented, stable, easy to use, heavily used by many repositories, licensed under a permissive license, platform neutral etc. I also contributed to the uthash repository many times and used it in other projects, so I am definitely biased. I agree that it might look complicated and over-designed for the purposes in MSL. But now, that we have it and distribute it and given that there was no other complaint so far, we can use it and I do not recommend to remove its usage just for the purpose.

DagBruck, 2020-05-28:
> I actually do like uthash.h as it is. It is well maintained, tested, documented, stable, easy to use, heavily used by many repositories, licensed under a permissive license, platform neutral etc. I [...]

Your view, whether biased or not, is also correct. But it worries me if something relatively simple as this drags in 1238 lines of foreign macro code, then something else will too. And the end result is overall code bloat. Also, in this case the code is pretty fragile, some other external code can make a `#define` and then all sorts of bad things happen.

It was perhaps wrong of me to point the finger specifically at UTHASH, but I worry about the overall effect.

sjoelund, 2020-05-28:
But nothing pulls in the `uthash.h` in headers, right? We only use the system headers and each file is compiled separate so there should be no actual problems (unless someone tries to compile all the ModelicaExternalC and user's external code by `#include` them all into one compilation unit). If any other header included it, we would have a problem.

```
$ grep -R uthash Modelica/Resources/C-Sources/ | cut -d: -f1 | uniq
Modelica/Resources/C-Sources/uthash.h
Modelica/Resources/C-Sources/ModelicaInternal.c
Modelica/Resources/C-Sources/ModelicaStandardTables.c
Modelica/Resources/C-Sources/ModelicaStrings.c
```

DagBruck, 2020-05-28:
Hmm, we have bad control over the compilation process. For example, Dymola has the possibility to export source-code FMUs and code for real-time platforms.

sjoelund, 2020-05-28:
Source-code FMUs are compiled unit by unit though. So it should still work in that case.

DagBruck, 2020-06-23:
I will close this issue because nobody seems particularly interested in removing this code dependency. No point in burdening the system with dead issues.

closed 2020-06-23 by DagBruck