modelica/ModelicaStandardLibrary #3569: Pressure drop in Modelica.Fluid.Pipes.DynamicPipe when dynamic momentum balance is taken into account

KarinHH, 2020-05-28:
I'm have a problem in understanding the simulation results of a discretized Modelica.Fluid.Pipes.DynamicPipe when using a compressible gas as medium and taking the dynamic momentum balance into account.
To illustrate that I built up a very simple model: pressure source + pipe + pressure sink. The pressure in the pressure source is linearly increased over time. The parameterization of the pipe mainly corresponds to the default values, but the parameter "momentumDynamics" is set to "Modelica.Fluid.Types.Dynamics.FixedInitial".

For lower gas velocities (=smaller inlet pressures) the pressure drop is somehow nearly linear distributed over the discrete elements of the pipe (of course the pressure drop is not the same in every element due to the change in medium properties). As the gas velocity gets higher however, the pressure drop in the last flow model (= resistive element) is dominating by far. Actually the when looking at the pressure distribution along the pipe it looks as if the pipe was choking. The picture below showes the pressures in the different flow models along the pipe. The pressure in the last flow model (green dashed line) corresponds to the constant pressure in the pressure sink.
[image]

This is however not possible since the velocities are still far below the velocity of sound. The velocity in the last flow model is a lot higher than in the rest of the pipe, because the pressure is a lot lower, since it corresponds to atmospheric pressure. This picture below shows the velocity in the flow models in the pipes as well as the velocities of sound. The velocities of sound are nearly constant at ~330 m/s.
[image]

What I do not understand:
Does the simulation result represents the physics correctly? If no, where is the "error" in the equations? If yes, what is the the physical behavior which the model represents here?

What I've tried:

 - Changing the discretization of the pipe does not change the phenomenom.
 - It seems to be independent of the medium model, I've also tried it with quite different medium models for compressible gas. (This example shown is using Modelica.Media.Air.ReferenceAir.Air_ph)
 - It only occurs if the dynamic momentum balance is chosen (**since despite the name of this flag, using this flag not only "activates" the dynamic term in the momentum balance but also adds the pressure loss due to acceleration**).

I've posted the same issue on stackoverflow, where I've got the hint to better try to file the issue directly at this place.

I've added the model as save-total.
[SimplePipe_pressureSourceTotal.zip](https://github.com/modelica/ModelicaStandardLibrary/files/4698300/SimplePipe_pressureSourceTotal.zip)

hubertus65, 2020-07-08:
First one comment: I think the pressure loss due to acceleration and the dynamic momentum balance should be separated. The pressure loss matters even for pure steady-state and gets significant in two-phase flows. But it arguably does not make sense to only use dynamic momentum balance and not the acceleration loss.

wischhusen, 2020-08-06:
Well, I believe it is an discretization error affecting the momentum balance. I can not explain in detial. But some lines seem strange to me. I thought that somehow the flow models are not staggered except the last one when looking at Wb_flows[n+1] for that case. 

Nevertheless, when trying all discretization options one finds that at least  modelStructure = av_vb will fail because the arrays of flowModel.m_flows and v_pipe1 do not match. I think some further analysis is required here.