modelica/ModelicaStandardLibrary #3570: Adding more reference papers and examples about how to define the flow/volume model.

Satinelamp, 2020-06-02:
I am learning how to build a thermo-fluid model with Modelica, I notice that in order to reduce the nonlinearity, it is recommended to use two different models: flow model and volume model, here is the explanation I found in a commercial library, but I am not sure how to arrange the equation in the components to realize this idea. So I suggest there should be some papers and examples about how to use this idea in the documentation.

[image]

hyumo, 2020-06-02:
This repo is mainly for Modelica Standard library development, I think your question is probably more appropriate for stackoverflow.

thorade, 2020-06-02:
There already is a related question on SO:
https://stackoverflow.com/questions/62141433/how-to-define-the-flow-model-and-volume-model-in-modelica-components

closed 2020-06-02 by Satinelamp