modelica/ModelicaStandardLibrary #3581: kg symbol of SIunits package

max-privato, 2020-06-19:
The icon of SIunits Package carries the text _kg_ as the unit of measure of mass.
However, it is written in Italic, while it should not.
I don't have here the text of ISO 31-1992 but the document "The International System of Units" from BIPM from which it derives. It says, in its 5.2 section:

**Unit names are normally printed in roman (upright) type**

Although they here use the word "normally", they don't discuss any possible exception, and throughout that document all the examples of units of measure are written  in roman (upright) style, not a single example is italic. So, we must not use _kg_ or _m/s_, always km or m/s.

So, I propose to use the correct symbol on our SIunits icon, i.e., with upright font style.

tobolar, 2020-06-19:
You are absolutely right concerning writing of units in a text.
I suppose the motivation for italic in that icon is to have similar look of icons there, see e.g. icon of `IconsPackage` or `TypesPackage` in MSL.Icons. In contrast, there is also `FunctionsPackage` having upright "f" - which definitely brakes the rule already.

henrikt-ma, 2020-06-22:
I like the idea of letting the icon convey more clearly that this is a package for units, by using upright font.

HansOlsson, 2020-06-23:
I think we could deviate from the SI-standard, but I don't see a problem with having "kg" either.
And if we use italic in icons I think we should do it consistently (e.g. italic for packages, and upright text for non-packages; except for info-classes?), and I don't see that logic at the moment, and in MSL 4.0.0 I see:

- Modelica.Units: hand-drawn italic
- Modelica.Units.SI: italic text
- Modelica.Icons.Function and Modelica.Icons.FunctionsPackage: upright text
- Modelica.Icons.IconsPackage: hand-drawn italic
- Modelica.Icons.TypesPackage: hand-drawn italic

The hand-drawn italic aren't tilted exactly the same angle as italic text (in Dymola at least), and except for the "kg" they are also non-straight lines. The 'i' for icon looks good, but honestly I don't think the 't' for TypesPackage is that good.

I believe there might be some issue with long texts in icon explaining why they are hand-drawn ( @DagBruck  ?), but I don't see a problem with these texts at the moment.

maltelenz, 2020-06-23:
Doesn't using actual text bring with it the problem of fonts, where not all environments and installations may have the font available (if it is even explicitly specified)? Hand drawing something avoids this problem altogether in cases where you really care about the exact look of something (like an icon).

HansOlsson, 2020-06-23:
> Doesn't using actual text bring with it the problem of fonts, where not all environments and installations may have the font available (if it is even explicitly specified)? Hand drawing something [...]

Possibly, but then "SI" and "f" has problems.

maltelenz, 2020-06-23:
> Possibly, but then "SI" and "f" has problems.

Yes, if we care this much about the look of our icons across tools and environments.

DagBruck, 2020-06-23:
Rendering of small text can be sensitive if the alignment is not perfect or the anti-aliasing kicks in and becomes too visible. This is an issue for very small icons. I think there are historical reasons too, in some cases maybe dating back to the last millennium.

tobolar, 2020-06-24:
> Hand drawing something avoids this problem altogether in cases where you really care about the exact look of something (like an icon).

Fine to me. But we shall define some basic rules (such as "font" height, inclination, etc.) to reach similar look of all letters used within icons. Esp. if the same letter apperas in various icons.

max-privato, 2020-06-27:
Nice to see that discussion of this ticket has gone soo deep.
I didn't realize that when we discuss whether to change a single icon, it is better to evaluate a general rearrangement of all text-based icons. But obviously this is the right thing to do.
I'll stay tuned.  Thank you for the time given to this ticket.

tobolar, 2020-10-05:
Summarizing the discussion here and in #3638 I suggest the following:

This concerns icons using only **one and two letters** as a symbol.
(for example: Modelica.Icons: Information, TypesPackage, FunctionsPackage, IconsPackage, TypeReal, TypeInteger, TypeBoolean, TypeString // Modelica.Units, Modelica.Units.SI, Modelica.Units.NonSI // Complex.'-', Complex.'*')
1. use text, not hand-drawn letters - it is much simpler to design an icon,
2. use standardized font: "serif" , "sans-serif" , and "monospace", see https://github.com/modelica/ModelicaStandardLibrary/pull/3638#issuecomment-703461139,
3. use size of text's graphical object (it shall be defined as well), not font size - to omit rendering problems with different sizes of icons,
4. use italic for packages and upright text for non-packages (except for info-classes), see https://github.com/modelica/ModelicaStandardLibrary/issues/3581#issuecomment-648093873.

Open question: how about symbols like '*'? Shall they be placed in the middle of icon?

henrikt-ma, 2020-10-05:
I understand that we need to take into consideration if there are some tools that have bugs in the implementation of a fixed font size, but apart from that I really think fixed font size is the way to go.  It is the only way to ensure consistent sizing of glyphs across icons.

maltelenz, 2020-10-05:
> I understand that we need to take into consideration if there are some tools that have bugs in the implementation of a fixed font size

I would say very little consideration should be taken for bugs in tools, if any. If the specification is unclear, we can avoid venturing into the unclear parts. However, if everyone agrees the specification is clear and the only issue is with a tool not implementing it correctly, tool bugs should not hold us back from choosing the best solution for the library.