modelica/ModelicaStandardLibrary #3582: Modelica.Icons.TypeReal not employed within SIUnits

AtiyahElsheikh, 2020-06-22:
I suppose that components within the Icons package imply that they are the community-recommended icons by MSL. 

Is there any specific reason against employing Modelica.Icons.TypeReal for the physical units within the SIunits package? 

Here is an example using Modelica.Icons.TypeReal:

`type Concentration = Modelica.Icons.TypeReal(final quantity="Concentration",final unit = "mol/l", min = 0);`

instead of the older way:

`type Concentration= Real(quantity="Concentration", unit="mol/l", min=0);`

HansOlsson, 2020-06-23:
I don't see a strong reason against, but I don't see a clear benefit either, and I see some minor issues.

In general the icon for a class to me has two obvious use-cases:

- In package browser, but in the SIunits package there would be 100+ units with the same icon, and thus it doesn't help to separate them from each other. In Modelica.Mechanics.MultiBody.Types and Modelica.Fluid.Types it at least helps to separate Reals from Integers (and we need something better for enumerations). Designing different units for different types will clearly be too much work.
- In models, but these variables normally don't have a graphical representation so no benefit in that case either (you cannot connect to them).

Thus it just seems to complicate the library for no clear benefit.

When a `Concentration `needs a graphical information it should often be a connector, and we still haven't solved how to combine units with graphical information for connectors in a simple way for Reals, see https://github.com/modelica/ModelicaSpecification/issues/921 and https://github.com/modelica/ModelicaSpecification/issues/2127 and it seems odd to first add graphical information to the Reals with unit that need it the least.

AtiyahElsheikh, 2020-06-23:
My current use case is as follows: 

The Biochem library have [some physical units]
(https://github.com/AtiyahElsheikh/BioChem/blob/develop/BioChem/Units/package.mo) some of which are already available in MSL. However they additionally have graphical annotation. My own library, GenKinetics, has also additional physical units. 

I feel like directly re-defining the units I need in terms of MSL.SIunits whenever available. So it becomes clear what is Biochem-specific / GenKinetics-specific and what belongs to MSL. However, they are already associated with the graphical annotations (which are also not important. 

In case nothing change I will extend from MSL.SIunits and keep the graphical annotations.

HansOlsson, 2020-12-02:
I don't see that this will be worked on for the foreseeable future. We might revisit after figuring out how to combine units and graphics for connectors in a simple way.

closed 2020-12-02 by HansOlsson