modelica/ModelicaStandardLibrary #3587: Reference trajectories of some variables in MSL 3.2.3 reference files only contain numerical noise

beutlich, 2020-06-26:
Folks, 

I'm afraid we still have the problem from #3476 in MSL 3.2.3 for the following test cases: 

`Modelica.Electrical.Machines.Examples.AsynchronousInductionMachines.AIMS_Start` [i_0_r](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Electrical.Machines.Examples.AsynchronousInductionMachines.AIMS_Start.diff.aims.i_0_r.html)
`Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_DOL` [i_0_s](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_DOL.diff.aimcE.i_0_s.html), [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_DOL.diff.aimcM.stator.zeroInductor.i0.html)
`Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_Transformer` [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_Transformer.diff.aimc.stator.zeroInductor.i0.html)
`Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_YD` [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_YD.diff.aimc.stator.zeroInductor.i0.html)
`Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_withLosses` [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_withLosses.diff.aimc.stator.zeroInductor.i0.html)
`Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SMPM_Braking` [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SMPM_Braking.diff.smpm.stator.zeroInductor.i0.html)
`Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Conveyor` [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Conveyor.diff.imc.stator.zeroInductor.i0.html)
`Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Initialize` [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Initialize.diff.imc.stator.zeroInductor.i0.html)
`Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Transformer` [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Transformer.diff.imc.stator.zeroInductor.i0.html)
`Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_YD` [i0](https://libraries.openmodelica.org/branches/newInst/Modelica_3.2.3/files/Modelica_3.2.3_Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_YD.diff.imc.stator.zeroInductor.i0.html)

@AHaumer, @christiankral, @beutlich, I guess the problem is that this change was applied to the 4.0.0 reference files (I don't see this problem in our [MSL 4.0.0 reports](https://libraries.openmodelica.org/branches/newInst/Modelica_4.0.0/Modelica_4.0.0.html)) but not to the 3.2.3 maintenance reference files, which was the original intent of #3476, as it is clear from the title 😃 

We should apply #3497 also to the maint/3.2.3 branch. @AHaumer, @christiankral, can you take care of that or shall I do it?

I guess we should also re-generate the reference files for MSL 3.2.3 accordingly. @beutlich could you take care of that? I could also run those examples in Dymola myself and only upload those files to GIT, but I guess re-running the scripts is much easier. 

Please let me know how we can proceed.

_Originally posted by @casella in https://github.com/modelica/ModelicaStandardLibrary/issues/3476#issuecomment-649867593_

christiankral, 2020-06-28:
I remember specifying the reference signals in files named `referenceSignals.txt`. The change was applied in #3497. Based on the specified reference signals, the `CSV` files were created by @beutlich in the MSL repository path `Modelica/Resources/Reference/`

I hope there is a straight forward way of creating the `CSV` files based on the `referenceSignals.txt` files and these changes had not to be applied manually. If that is the case, it shall be sufficient to specify the `referenceSignals.txt` files in https://github.com/modelica/MAP-LIB_ReferenceResults/tree/v3.2.3/Modelica/ as this seems to be the correct location. 

@beutlich could you then create the matching reference `CSV` files?

casella, 2020-06-28:
@christiankral, I can back-port the changes to referenceSignals.txt the maint/3.2.3 branch if that is ok for you.

@beutlich, @GallLeo  I would gladly take care of the generation of the files, but I need some references on the procedure. Is there any howto/readme/guideline about that?

beutlich, 2020-06-29:
> @beutlich could you then create the matching reference `CSV` files?

Can do.

> Is there any howto/readme/guideline about that?

Yes, there is a dedicated [wiki page](https://github.com/modelica/ModelicaStandardLibrary/wiki/Regression-testing-of-Modelica-and-ModelicaTest) which also is linked from the wki landing page: https://github.com/modelica/ModelicaStandardLibrary/wiki. The regression testing was documented for MSL v4.0.0. For MSL v3.2.x you need to take the comparisonSignals.txt from https://github.com/modelica/MAP-LIB_ReferenceResults since they are not part of the MSL.

christiankral, 2020-06-29:
OK, I created the PR https://github.com/modelica/MAP-LIB_ReferenceResults/pull/2 but I cannot assign any reviewer or lables... Some else with more privileges, e.g. @GallLeo, has to do this.

beutlich, 2020-06-29:
> but I cannot assign any reviewer or lables... Some else with more privileges, e.g. @GallLeo, has to do this.

Fixed. I updated the access rights from Read access to Triage access for all members of @modelica/map-lib.

beutlich, 2021-01-23:
Since https://github.com/modelica/MAP-LIB_ReferenceResults/pull/2 is merged, the reference results are cleaned up whenever a new MSL 3.2.3 build is scheduled.

closed 2021-01-23 by beutlich