modelica/ModelicaStandardLibrary #3601: Support for n-Dimensional arrays in IO functions using HDF5 file format

alkhwarizmi, 2020-07-07:
Dear all,

it seems that support for 3D arrays and, in general, n-dimenisonal arrays is missing in the MSL (see discussion [here](https://trac.openmodelica.org/OpenModelica/ticket/6046#comment:4)). In particular I am referring to the readArraySize() and read3DArrayFromFile() functions.

Although full support of arrays with an a priori unknown number of dimensions may require additions to the language itself, and may be regarded as unlikely to happen, a limited support for 3D, 4D, 5D (the highest number to be arbitrarily decided) arrays could be added just extending the library. For example, my current application would need functions like readArraySize(fileName:String, arrayName:String):1DArray and read3DArrayFromFile(fileName:String, arrayName:String, dims:1DArray):3DArray

Moreover, I propose to start building a IO mechanism to load and save n-dimensional arrays or a cluster of them relying on the HDF5 standard (the idea is that any HDF5 reader could open the data file). I don't think MSL should support the entire standard, but just the subset of it required to support its own data structures, including n-dimensional arrays.

In the following months I could be in the position to contribute something with this regard if somebody can let me know where to start.

Sincerely,

beutlich, 2020-07-09:
I'd claim your issue fits better to https://github.com/modelica-3rdparty/ExternData.

alkhwarizmi, 2020-07-10:
Thank you for the information. I was not aware of the library and that could be a good point to solve my contingent problems. However, I don't fully agree that the question does not fit this project.  It would if MSL provided to the users at least one standard method to save and load n-dimensional arrays on and from disk but, as far as I know, this is not possible in the current library (without workarounds like saving a flattened version of the arrays and the meta information needed to reconstruct them). I perceive this as a failure of the MSL. I am wondering which language supporting n-dimensional arrays does not provide the user with the infrastructure to save and load them from file.