modelica/ModelicaStandardLibrary #3603: Memory allocation error in CombiTimeTable when given UTF16 LE file

maltelenz, 2020-07-08:
If a CombiTimeTable is given a UTF16-LE encoded text file of sufficient length (pretty short) it fails with a `Memory allocation error`.

I realize the table format requires UTF-8 or ASCII, but a nicer failure mode for such an invalid file would be nice.

To reproduce, create a model with a single CombiTimeTable component pointing to the attached file, and attempt to simulate:

[MadeUpTable.txt](https://github.com/modelica/ModelicaStandardLibrary/files/4890651/MadeUpTable.txt)

The error seems to come from https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Resources/C-Sources/ModelicaIO.c#L1039 after `bugLen` overflows to negative numbers.

beutlich, 2020-07-09:
@maltelenz Do you think there might be some requirement to support UTF-16 encoded text files?

closed 2020-07-09 by beutlich

maltelenz, 2020-07-09:
@beutlich I don't see a large need for it. My personal experience is that everyone is using UTF-8, and the description on https://en.wikipedia.org/wiki/UTF-16 seems to agree:

> It is rarely used for files on Unix-like systems. As of May 2019, Microsoft seems to have reversed course and now supports and recommends using UTF-8.[2]

> UTF-16 never gained popularity on the web, where it is used by under 0.01% (1 hundredth of 1 percent) of web pages,[3] and UTF-8 dominates.

PR #3604 (2020-07-09): ModelicaIO.c: Improve error message if reading text file containing zero bytes
closes #3603