modelica/ModelicaStandardLibrary #3605: Add more display unit for the SpecificEnergy and SpecificEnthalpy type

Satinelamp, 2020-07-09:
There should be more display units for the SpecificEnergy and SpecificEnthalpy type, for example, "kJ/kg", "MJ/kg", cause in the thermo-fluid model, the steam enthalpy is often at the scale of "kJ/kg", it will be easier to understand the result with new display units.

[image]

[image]

beutlich, 2020-07-09:
I believe this is a tool-issue and nothing that can be fixed in the library.

[image]

closed 2020-07-09 by beutlich

Satinelamp, 2020-07-09:
Could you tell me what the dialogue window is in your screenshot?

HansOlsson, 2020-07-09:
> Could you tell me what the dialogue window is in your screenshot?

That dialogue must be from a non-Dymola tool. 

You seem to be using Dymola and the procedure for adding display units in Dymola is described in the 1.3.17 Display units in 
Dymola 2021\Documentation\Dymola User Manual 1B.pdf

Please contact your normal support channel if you need more information.

beutlich, 2020-07-09:
> Could you tell me what the dialogue window is in your screenshot?

That was from SimulationX. It has a large built-in database for (display) unit conversion, which however is not extensible by users.