modelica/ModelicaStandardLibrary #3606: CombiTable1Ds do not infer table size correctly from file

dariomangoni, 2020-07-09:
CombiTable1Ds output size is always 1 even if the table **contained in the file** has >2 columns.
Passing the table as a matrix (i.e. not from external file) does work.

In the attachment look at `CombiTable1DsTXT_multicol_space_NoOutput` to have a compilable example that shows the issue.

Please double check if I didn't put any mistake in the loading files.

Various compilers shows the same issue thus suggesting an error arising from the base code.

[CombiTableExternalData.zip](https://github.com/modelica/ModelicaStandardLibrary/files/4895755/CombiTableExternalData.zip)

beutlich, 2020-07-09:
Yes, this is a known design flaw. (I tried to remember where I answered it first, but it seems that was a private conversation only.)

When you set tableOnFile=true, you basically deactivate/disable the table matrix parameter. However, it still is referred in the columns parameter, i. e. columns=2:size(table,2). Since the output size must be known at translation time, you can either set

1. table=fill(0.0, 0, 3) or
2. columns=2:3

to adapt it according to the actual number of columns read from file..

closed 2020-07-09 by beutlich

dariomangoni, 2020-07-09:
Well, it's a strange way to close an issue.
At least provide appropriate warning in the documentation if not solving the issue.