modelica/ModelicaStandardLibrary #3607: problem with the simulation of Implement idealized battery #2957

saer1994, 2020-07-15:
I' m currently Dymola 2012 FD01 with MSL 3.2, when I try to simulate "BatteryDischargeCharge" (the example model) the following error occurs: "Error: Class or component 'extrapolation' not found in CombiTable1Ds combiTable1Ds". How can I fix this error? I already check if I have in the MSL the component Blocks.Types.Extrapolation ( I have this component but in the code of idealized battery there is the following assignment:  extrapolation=Modelica.Blocks.Types.Extrapolation.NoExtrapolation).
What is NoExtrapolation?

saer1994, 2020-07-15:
@AHaumer

AHaumer, 2020-07-15:
The problem is:
You try to simulate a model from MSL 4.0.0 with MSL 3.2.
The CombiTable1Ds has been enhanced in between, adding a new parameter extrapolation.
Stay consistent, use the model with MSL 4.0.0. You can download MSL 4.0.0 from github. 
I suppose you did that, otherwise you won't have the BatteryDischargeCharge model.
BTW: Your Dymola version is pretty old (8 years). Using a more up-to-date version (2021) is rcommended.

saer1994, 2020-07-15:
I have to work with this version because I have to use my  model for an H.I.L test. Where I can find the update of  CombiTable1Ds' s code? could you please attach the code?

AHaumer, 2020-07-15:
No I can't attach the code - that will end up in a mess. You can open MSL 4.0.0 with Dymola 2012 FD01. I suppose you need the correspondig C-code for the tables. Maybe @beutlich can help you - he knows the tables very well.

I can't remember the behaviour of the "old" CombiTable1ds - what happens if the abscissa leaves the defined range?
Anyhow, extrapolation below SOC=0 and above SOC=1 is not meaningful. You could try to remove the parameter propagation of "extrapolation", and insert an assert-statement in the battery model when SOC leaves the defined range.

saer1994, 2020-07-15:
I solve the issue removing the extrapolation parameter, thank you vary much for your avaibility

AHaumer, 2020-07-17:
Thanks for the feedback

closed 2020-07-17 by AHaumer