modelica/ModelicaStandardLibrary #3610: shapeType documentation example for file:// is inconsistent with standard file URIs

maltelenz, 2020-07-23:
The standard `file://` URI scheme usually has three front slashes even on Windows, see https://en.wikipedia.org/wiki/File_URI_scheme

> Or for a local file, the hostname is omitted, but the slash is not (note the third slash): `file:///c:/path/to/the%20file.txt`

However, the MSL documentation only has two slashes in the example:

https://github.com/modelica/ModelicaStandardLibrary/blob/4929e9736444bc20a20a3dc3c3767886a8bf0904/Modelica/Mechanics/MultiBody/Visualizers/FixedShape.mo#L94-L96

Should we change so we use proper `file` URIs?

HansOlsson, 2020-07-23:
Yes, using proper file-URIs here would make sense.
(We also have a separate issue of modifying the modelica-URIs.)

maltelenz, 2020-07-23:
> (We also have a separate issue of modifying the modelica-URIs.)

Right, I didn't mention that since I understand there is larger work underway of redesigned those.

henrikt-ma, 2020-08-04:
But this looks like it really is supposed to be a file URI, not a Modelica URI, right?  To me it seems as if we should simply fix it the way @maltelenz suggests (and I can find four occurrences to fix).

beutlich, 2020-08-04:
Feel free to create a PR for master branch. Thanks.

PR #3613 (2020-08-04): Add missing URI separator in file URIs
Fixes #3610
commit 2dc388ba (2020-08-04): Add missing URI separator in file URIs (documentation only) (#3613)

Fixes #3610
closed 2020-08-04 by beutlich