modelica/ModelicaStandardLibrary #3622: Unused parameter in function

casella, 2020-08-29:
This parameter

https://github.com/modelica/ModelicaStandardLibrary/blob/39285d3bc28bc947613db51aff015f7ccd35b96f/Modelica/Electrical/Machines/SpacePhasors/Functions/ToSpacePhasor.mo#L14

shows up in function [toSpacePhasor](https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Electrical/Machines/SpacePhasors/Functions/ToSpacePhasor.mo), but is not used in the function body. 

@AHaumer, @christiankral, why was it defined in the first place?

adrpo, 2020-08-29:
I think is just a copy paste mishap, if you look at:
https://github.com/modelica/ModelicaStandardLibrary/blob/39285d3bc28bc947613db51aff015f7ccd35b96f/Modelica/Electrical/Machines/SpacePhasors/Functions/FromSpacePhasor.mo#L13
there the parameter: `TransformationMatrix` is not used.

adrpo, 2020-08-29:
See #3623

christiankral, 2020-08-31:
The obsolete code fragments are leftovers from creating functions/inverse functions and blocks/inverse blocks from the same code base. So yes, the obsolete code shall be removed.

christiankral, 2020-08-31:
See https://github.com/adrpo/Modelica/pull/1

closed 2024-01-18 by Harisankar-Allimangalath