modelica/ModelicaStandardLibrary #3630: Unit present verbatim in text macro

maltelenz, 2020-09-23:
In `Modelica.Blocks.MathBoolean.OnDelay`, there is a text macro `%delayTime s` on the icon view, at https://github.com/modelica/ModelicaStandardLibrary/blob/39285d3bc28bc947613db51aff015f7ccd35b96f/Modelica/Blocks/MathBoolean.mo#L414
Because of https://github.com/modelica/ModelicaSpecification/issues/2339 , the unit should be removed from the macro (tools are expected to add it).

beutlich, 2020-09-23:
MO-2339 was decided, but later closed without being added to MLS 3.5.

maltelenz, 2020-09-23:
The part about showing displayunit was added in https://github.com/modelica/ModelicaSpecification/commit/ddbaf0847f330bd09eb0f3ed039aff27ed51e760 ? If I'm reading it right, only the part about significant digits was not added.

MartinOtter, 2020-12-15:
I agree that this should be changed. However, currently have no time to do it.