modelica/ModelicaStandardLibrary #3647: Linker warning mktemp is dangerous

AnHeuermann, 2020-10-08:
## Description
When compiling Modelica models which use ModelicaMatIO the linker will issue warnings like:
```bash
/usr/bin/ld: /home/andreas/workspace/OpenModelica/build/bin/../lib/x86_64-linux-gnu/omc/libModelicaMatIO.a(ModelicaMatIO.o): in function `Mat_VarDelete':
/home/andreas/workspace/OpenModelica/OMCompiler/3rdParty/ModelicaExternalC/BuildProjects/autotools/../../C-Sources/ModelicaMatIO.c:4854: warning: the use of `mktemp' is dangerous, better use `mkstemp' or `mkdtemp'
```
In this case OpenModelica was used as generation tool, but unless other tools don't suppress warnings this should be relevant for them as well.
This warning could discomfort users that have models that are running well and suddenly start generating warnings, depending on their used linker.

Those warnings originate from ModelicaMatIO which is using `mktemp` (and `tmpnam` as well).

## Related issues
https://github.com/modelica/ModelicaStandardLibrary/issues/2292
[OpenModelica Trac ticket #6152](https://trac.openmodelica.org/OpenModelica/ticket/6152)

## 
OS: Ubuntu 20.04
MSL Version: 3.2.3

HansOlsson, 2020-10-08:
The code is common for VS under Windows, and gcc; but any alternative would need to be different - and it seems it would require changes in interfaces as well.

Note that there seems to be two parts of the danger:

- Some other process creating the same file-name during that time-interval by mistake, but that risk is very low if mktemp has some reasonable logic (including process id).
- An attacker doing this, i.e. a security risk. If you are relying on this function in a Modelica model on system that has security implications and attackers have access to the temp-directory I think your system needs a more complete security overview.

I don't know why that function uses a temporary file at all; but avoiding a temporary file seems the best solution.

AnHeuermann, 2020-10-08:
Thanks for the fast response.

For me hiding warnings from users they can't do anything about would be most important.
Of course getting a more secure program would be a nice bonus.

beutlich, 2020-10-11:
Despite being a non-issue (since Mat_VarDelete is never called), it is fixed in the upstream [matio release 1.5.18](https://sourceforge.net/p/matio/news/2020/09/version-1518-release/) and will get fixed for MSL once #3641 is merged.

closed 2020-11-18 by beutlich