modelica/ModelicaStandardLibrary #3650: Modelica Media Interface is information incomplete

dzimmer, 2020-10-19:
The Modelica Media package provides an abstract interface to the thermodynamic state of the Medium by defining a set of "set" and "get" like functions. So far, so good.

Unfortunately it occured to me that this interface is incomplete. Using the "get"-like functions, I cannot fully retrieve the thermodynamic state because there seems to be no function for the (independent) mass fractions. When you start using the Media models with different interfaces than in Standard fluid, you start to notice.

Irrespective of what a modeller wants to do, I think it is a good idea to make the abstract interface complete. It can also be easily fixed. 

I propose to add the following function to PartialMedium

`  
  replaceable partial function massFraction "returns independent mass fraction"
    extends Modelica.Icons.Function;
    input ThermodynamicState state "Thermodynamic state record";
    output MassFraction Xi[nXi] "(independent) Mass Fraction";
  end massFraction;
`

Then add to PartialPureSubstance the implementation

`  redeclare replaceable function massFraction "Return independent mass Fraction"
    extends Modelica.Icons.Function;
    input ThermodynamicState state "Thermodynamic state record";
    output MassFraction Xi[nXi] "(independent) Mass Fraction";
  algorithm 
    Xi := fill(0,0);
  end massFraction;`

and last to PartialMixtureMedium

`  replaceable function massFraction "Return independent mass Fraction"
    extends Modelica.Icons.Function;
    input ThermodynamicState state "Thermodynamic state record";
    output MassFraction Xi[nXi] "(independent) Mass Fraction";
  algorithm 
    Xi := state.X[1:nXi];
  end massFraction;
`

Objections? Other ideas? Shall I branch and raise a merge request?

We need this for our new fluid library.

beutlich, 2020-10-26:
@thorade @harmanpa Did you also notice when designing your libraries?

@hubertus65 Can we consider this as a short-coming of current Modelica.Media? Did @dzimmer mentioned all partial packages where massFraction shall be added? See also your diagrams in #85.

harmanpa, 2020-10-26:
> @thorade @harmanpa Did you also notice when designing your libraries?

For ClaytexFluid and FluidPower we moved away from Modelica.Media/Fluid as we had very specific requirements that could be satisfied with a more compact design.

thorade, 2020-10-26:
I only implemented pure fluids, so never had a need to get the mass fractions.
To me, the suggestion looks reasonable, no objections.

MartinOtter, 2020-12-15:
Yes, should be included. @dzimmer, please provide a pull request

hubertus65, 2021-01-20:
I agree with the suggestion, it looks reasonable. One possible issue with this is that for many applications, mole fractions are much more suitable than mass fractions, and others again prefer extensive properties for everything with mixtures (Moles instead of mole fractions, Molar flows in interfaces). But as a fix for what we have now, this works for me and is for sure an improvement.

closed 2024-01-09 by hubertus65