modelica/ModelicaStandardLibrary #3651: complex transfer function block uses wrong order of coefficients

HoWilgh, 2020-10-21:
The coefficients of the denominator of the complex tranferFunction block (Modelica.ComplexBlocks.ComplexMath.TransferFunction) are used in the wrong order.
The information next to the input field of the 'Parameters' 'a' and 'b' reads 'Denominator coefficients of transfer function (e.g., 5*s+6 is specified as {5,6})' but it is the other way around. One has to insert {6,5} for the above example!
The model 'ShowTransferFunction "Test Complex Transfer Function Block"' does look like it works but it does not. This is because the polynomial coefficients of the transfer function 'a' where chosen in a 'symmetric' way 'parameter Real a[:]={1,2*d,1}'. One has to change one of the two '1's to see the effect!

I suggest a modified version of the model with more realistic chosen parameters like:
``
model ShowTransfe
rFunction "Test Complex Transfer Function Block"
  extends Modelica.Icons.Example;
  parameter Real d=0.01  "Damping coefficient in kg/s (not the damping ratio)";
  parameter Real m=0.1 "mass in kg";
  parameter Real c=0.1 "stiffness in N/m";
  
  
  parameter Real b[:]={-m} "Numerator polynomial coefficients of the transfer function";

  parameter Real a[:]={c,d,m} "Denominator polynomial coefficients of the transfer function
  a = c + d*s + m*s^2";

  parameter Real wMin=0.01 "Lower bound for frequency sweep";
  parameter Real wMax=100 "Upper bound for frequency sweep";
  Real lg_w=log10(logFrequencySweep.y) "Logarithm of frequency";
  Real dB=20*log10(complexToPolar.len) "Magnitude of the transfer function in decibel";
  Modelica.SIunits.Angle phi(displayUnit="deg")=complexToPolar.phi "Argument of the transfer function";
  Modelica.ComplexBlocks.Sources.LogFrequencySweep logFrequencySweep(
    duration=1,
    wMin=wMin,
    wMax=wMax)
    annotation (Placement(transformation(extent={{-80,-40},{-60,-20}})));
  Modelica.ComplexBlocks.Sources.ComplexConstant const(k(re=1, im=0))
    annotation (Placement(transformation(extent={{-80,-10},{-60,10}})));
  Modelica.ComplexBlocks.ComplexMath.TransferFunction transferFunction(b=b,
      a=a)
    annotation (Placement(transformation(extent={{-40,-10},{-20,10}})));
  Modelica.ComplexBlocks.ComplexMath.ComplexToPolar complexToPolar
    annotation (Placement(transformation(extent={{0,-10},{20,10}})));
equation
  connect(const.y, transferFunction.u)
    annotation (Line(points={{-59,0},{-50.5,0},{-42,0}}, color={85,170,255}));
  connect(logFrequencySweep.y, transferFunction.w)
    annotation (Line(points={{-59,-30},{-30,-30},{-30,-12}}, color={0,0,127}));
  connect(transferFunction.y, complexToPolar.u)
    annotation (Line(points={{-19,0},{-2,0}}, color={85,170,255}));
  annotation (
    experiment(StopTime=1, Interval=0.001), Documentation(info=
               "<html>
<p>This example shows the response of a PT2 (mechanical spring-mass-damper- system with an acceleration acting on the mass) defined by its transfer function</p>
<pre>
             -m
H(jw)=-------------------
      c + d* jw + m*(jw)^2
</pre>
<p>Frequency performs a logarithmic ramp from 0.01 to 100 s^-1.</p>
<p>
Plot the magnitude locus (in dB) dB versus lg_w and the phase locus versus lg_w.
</p>
</html>"));
end ShowTransferFunction;
`
BR,
Wilfried

beutlich, 2021-01-09:
@AHaumer Is is just this example `ShowTransferFunction` model or the underlying block `TransferFunction` itself?

christiankral, 2021-01-23:
The time transient transfer function `Modelica.Blocks.Continuous.TransferFunction` shows the documentation:

```
        b[1]*s^[nb-1] + b[2]*s^[nb-2] + ... + b[nb]
y(s) = --------------------------------------------- * u(s)
        a[1]*s^[na-1] + a[2]*s^[na-2] + ... + a[na]
```

The documentation of the complex transfer function `Modelica.ComplexBlocks.ComplexMath.TransferFunction` shows:

```
        b[1]*(jw)^[nb-1] + b[2]*(jw)^[nb-2] + ... + b[nb]
y(jw) = ------------------------------------------------- * u(jw)
        a[1]*(jw)^[na-1] + a[2]*(jw)^[na-2] + ... + a[na]
```

Therefore, the documentation suggests the same meaning of the coefficients. I propose that we keep the documentation and update the implemented code of `Modelica.ComplexBlocks.ComplexMath.TransferFunction` as this seems to be bug. Otherwise we provide two different implementations with different meanings of the polynomial coefficients in the MSL with I am definitely not in favor of.

I will prepare a PR.

christiankral, 2021-01-24:
@HoWilgh You proposed the parameters 
```modelica
  parameter Real d=0.01 "Damping coefficient in kg/s (not the damping ratio)";
  parameter Real m=0.1 "Mass in kg";
  parameter Real c=0.1 "Stiffness in N/m";
```
I propose to go for a slightly different set of parameters as your parameter combination again result in a "symmetrical" vector `{0.1,0.01,0.1}`. I'd prefer to avoid ambiguities again. So how about `m = 0.2` instead?

closed 2022-10-07 by MartinOtter