modelica/ModelicaStandardLibrary #3653: Internal comments left over in Modelica.Media.Water.IF97_Utilities.BaseIF97.Regions.{hl_p_R4b, sv_p_R4b}

beutlich, 2020-10-26:
There are internal comments left over that should be either removed or improved.

https://github.com/modelica/ModelicaStandardLibrary/blob/aa92ae0c2c7289d27ca6b746ea66d20b9d41d053/Modelica/Media/Water/IF97_Utilities.mo#L797-L798

and

https://github.com/modelica/ModelicaStandardLibrary/blob/aa92ae0c2c7289d27ca6b746ea66d20b9d41d053/Modelica/Media/Water/IF97_Utilities.mo#L847-L848.

If the referenced notebook R3Approx.nb still is available it would be best to add it as PDF to the documentation of the functions [hl_p_R4b](https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_Media_Water_IF97_Utilities_BaseIF97_Regions.html#Modelica.Media.Water.IF97_Utilities.BaseIF97.Regions.hl_p_R4b) and [sv_p_R4b](https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_Media_Water_IF97_Utilities_BaseIF97_Regions.html#Modelica.Media.Water.IF97_Utilities.BaseIF97.Regions.sv_p_R4b).

hubertus65, 2020-10-26:
Interesting note! Thinking about when I implemented this, this must be approximately from the summer of 2000, in my own Thermofluid Library, a bit over 20 years ago. This will be a challenge to dig up. I suggest this is best to remove the comments if I don't get back to this within a few weeks.

hubertus65, 2020-10-27:
I had no idea that I did take care of my old files this well: I found it and will attach both a pdf and a 
Mathematica Notebook, for anyone who can and wants to dig into the details. I could not attach the Notebook as is, only zipped, if somebody wants it. Note it was renamed for more clarity. 
[IF97R3Approx.zip](https://github.com/modelica/ModelicaStandardLibrary/files/5447855/IF97R3Approx.zip)
[IF97R3Approx.pdf](https://github.com/modelica/ModelicaStandardLibrary/files/5447842/IF97R3Approx.pdf)

beutlich, 2020-10-27:
Nice. Thanks. I believe we could add it to the Resources/Documentation/Media/Water/IF97documentation and link to it. I hope it will be fine to use it though it is GPL.

beutlich, 2020-10-27:
Do you also have the referenced satp.dat file?

hubertus65, 2020-10-28:
I think I have the files and can add them. BTW: no problem to relicense it as BSD for me, I just can't edit the .nb-files any more since I don't have a Mathematica license.

hubertus65, 2020-10-28:
Here are the data files for the saturation properties that I used. They are from a C-code implementation that I did from the same code-generation (from Mathematica) that eventually became the Modelica-code. 
[if97r3.zip](https://github.com/modelica/ModelicaStandardLibrary/files/5454879/if97r3.zip)