modelica/ModelicaStandardLibrary #3655: ModelicaServices.ExternalReferences.loadResource violates purity rules

henrikt-ma, 2020-11-02:
The demo implementation of `ModelicaServices.ExternalReferences.loadResource` isn't marked impure,

https://github.com/modelica/ModelicaStandardLibrary/blob/81dd1b63a61075c85fbd26945eb333b202b9e4b9/ModelicaServices/package.mo#L164

Yet it calls the impure `Modelica.Utilities.Files.fullPathName`,

https://github.com/modelica/ModelicaStandardLibrary/blob/81dd1b63a61075c85fbd26945eb333b202b9e4b9/Modelica/Utilities/Files.mo#L568

Even though this really just concerns a dummy implementation of ModelicaServices, I'd interpret the purity of `loadResource` as dictating how it is also supposed to be handled in actual implementations of ModelicaServices.  For example, marking it `impure` clearly shows that a call to `loadResource` isn't allowed in a constant's declaration equation.

henrikt-ma, 2020-11-02:
@maltelenz just pointed out to me that not having `loadResource` marked `impure` makes the deprecated semantics kick in, where the impure function can actually be used to initialize a constant:
> A deprecated semantics is that external functions (and functions defined in Modelica directly or indirectly calling them) without `pure` or `impure` keyword are assumed to be impure, but without any restriction on calling them. Except for the function `Modelica.Utilities.Streams.print`, diagnostics must be given if called in a simulation model.

This means that `loadResource` would need to be `pure` in order to be backwards compatible, suggesting that when called with a constant argument, the call could be evaluated already during translation.  I'm not sure this counts as being backwards compatible in a good way…

HansOlsson, 2023-09-05:
The purity rules are in general messy and might need further revisions, and loadResource is no exception.

In one sense loadResource has impure semantics, as it has a side-effect in terms of referencing the external resource (and tools do external changes based on that). In another sense the intent inside the model is that the resource is just there and not changed by loadResource. But then there's the fact that it is normally used together with reading the referenced external resource and that is less pure - so I'm not sure if marking it as impure would be _that_ problematic, but doing it just before a minor revision is far from ideal.

As for fullPathName it is impure in the sense that it under Windows depends on the current directory (and possibly some other settings), and on under Linux additionally follows symbolic links.