modelica/ModelicaStandardLibrary #3660: Modelica.Utilities.Files.createDirectory fails if the directory does not exist

sjoelund, 2020-11-11:
`Modelica.Utilities.Files.createDirectory` calls `fullPathName`, which fails if the path does not exist (on Linux). This means createDirectory does not work. Some possible solutions:

* Change `fullPathName` to never fail (revert to the old code with `getcwd`)
* Change `createDirectory` to not call `fullPathName` (but this would not fix older MSL versions)
* Add an argument to `fullPathName` to not fail (return the input perhaps), and create this additional file in the external sources

beutlich, 2020-11-11:
Bug has been around for 7 years. If it is okay for you, I think we should revert 75d86f36e37c905db1c10f2d69b58324a11df1db (the realpath part) then.

sjoelund, 2020-11-11:
I think it would be nicer to just see if the result is NULL and then use the old code. That shouldn't be very difficult to add. Otherwise the functionality is also very different in Linux and Windows since ../ would never be replaced

PR #3663 (2020-11-17): Add getcwd fallback in ModelicaInternal_fullPathName
Check if realpath fails for non-existing path and add getcwd fallback. Especially, do not call ModelicaError for non-existing path.

Closes #3660 for OM#6204.
PR #3667 (2020-11-18): maint/4.0.x: Add getcwd fallback in ModelicaInternal_fullPathName
Back-port #3663 from master to fix #3660 in maint/4.0.x.
PR #3668 (2020-11-18): maint/3.2.3: Add getcwd fallback in ModelicaInternal_fullPathName
Back-port #3663 from master to fix #3660 in maint/3.2.3.
closed 2020-11-17 by beutlich