modelica/ModelicaStandardLibrary #3675: Search on doc.modelica.org not fully working

HansOlsson, 2020-11-26:
The search was added in #3579 - but it doesn't seem to work reliably.

Typing "clutch" into the search field finds nothing for Dymola and MapleSim, sort of works for Wolfram System Modeler and fully for OpenModelica. Some other keywords finds thing, I don't know if it always works for OpenModelica or not.

I verified that https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_Mechanics_Rotational_Components.html#Modelica.Mechanics.Rotational.Components.Clutch and 
https://doc.modelica.org/Modelica%203.2.3/Resources/helpMapleSim/Mechanics/Rotational/Components/index.html#Clutch
do exist.

So it seems it is a misconfiguration in some way, or those help-pages do in some way not conform to algolia requirements.

beutlich, 2020-11-26:
I also noticed and already reported it at https://github.com/algolia/docsearch-configs/issues/2753.

beutlich, 2020-11-26:
However, "clutch" works for me, see

[image]

HansOlsson, 2020-11-26:
Hmmm... It also works for me now, could be that it updated.

beutlich, 2020-12-13:
@HansOlsson OK for you to close?

HansOlsson, 2020-12-14:
> @HansOlsson OK for you to close?

Yes-

closed 2020-12-14 by HansOlsson