modelica/ModelicaStandardLibrary #3680: Quality of DynamicPipeClosingValve reference result

henrikt-ma, 2020-12-07:
A quick look at `ModelicaTest.Fluid.TestPipesAndValves.DynamicPipeClosingValve` doesn't reveal anything that has changed in this simple model from version 3.2.3 to 4.0.0.  Yet, the reference result look rather different, and also qualitatively different from what I get with System Modeler:
[image]

Has something really changed in the model from 3.2.3 to 4.0.0.  Can the reference result be considered reliable?

beutlich, 2020-12-07:
Telling from #3415 I remember that this particular model has trouble with solver settings in Dymola. It did not work for the default tolerance.

beutlich, 2021-05-21:
This issue den not seem to be connected with #3778 as `simulateModel("ModelicaTest.Fluid.TestPipesAndValves.DynamicPipeClosingValve", stopTime=3, numberOfIntervals=5000, tolerance=1e-05, resultFile="DynamicPipeClosingValve");` still does not terminate when tested in Dymola 2022.