modelica/ModelicaStandardLibrary #3687: Functions returning uninitialized fields

HansOlsson, 2020-12-15:
From https://github.com/modelica/ModelicaSpecification/issues/2745

In Modelica.Media.Water.IF97_Utilities.dewcurve_p either bpro.cv or bpro.cp is returned as uninitialised variable.
The uninitialized field isn't used further on, but it isn't ok.

Similarly Modelica.Mechanics.MultiBody.Visualizers.Advanced.SurfaceCharacteristics.torus returns a zero-sized uninitialized array C.

thorade, 2020-12-15:
Is there a way to detect uninitialized variables systematically?

beutlich, 2020-12-15:
The Media issue is a duplicate of #1910. I guess we can focus on the MultiBody case here.

HansOlsson, 2020-12-15:
> Is there a way to detect uninitialized variables systematically?

It seems it is being/was added in Wolfram SystemModeler (I think), and I'm pretty confident that other tools will add it as well.

tobolar, 2021-10-04:
This is probably also an issue for `Mechanics.MultiBody.Visualizers.Advanced.SurfaceCharacteristics.rectangle`.

Simply solution `C := zeros(0,0,3);` ?