modelica/ModelicaStandardLibrary #3695: Trajectory of 'k' in R134a_setState_pTX doesn't look suitable for comparison

henrikt-ma, 2020-12-22:
Can someone please verify that `k` is a meaningful comparison signal in `R134a_setState_pTX`:
- https://github.com/modelica/MAP-LIB_ReferenceResults/blob/v4.0.0/ModelicaTest/Media/TestOnly/R134a_setState_pTX/comparisonSignals.txt

It doesn't look like the typical comparison signal:
[image]

beutlich, 2020-12-22:
Negative density does not look either.

wischhusen, 2020-12-23:
The issue is the iterative calculation of d from p and T. In some parts of the liquid region the algorithm in dofpt fails to find the correct solution.

beutlich, 2020-12-23:
Thank you for confirmation. I agree that changing the iteration

```diff
--- a/Modelica/Media/R134a.mo
+++ b/Modelica/Media/R134a.mo
@@ -2510,7 +2510,7 @@ This function integrates the derivative of temperature w.r.t. time in order to a
         liquid := T <= Modelica.Media.R134a.R134a_ph.saturationTemperature(p);
         if liquid then
           d := R134aData.data.FDCRIT*Common.CubicSplineEval(localx, dl_coef[int,
-            1:4])*1.02;
+            1:4])*1.22;
         else
           d := R134aData.data.FDCRIT*Common.CubicSplineEval(localx, dv_coef[int,
             1:4])*0.95;
```

resolves the issue __for this test model__. But that changed scaling parameter seems arbitrary. Do you/we have a dedicated R134a testsuite to check for regressions?

wischhusen, 2021-01-06:
I would like to propose the following solution. I changed the guess values by using the temperature interpolation for boundary states of density. This leads to stable iterations for the tester. I also tried pressure slopes in gas, supercritical and liquid region.

function dofpT
  extends Modelica.Icons.Function;
  input Modelica.SIunits.Pressure p "Pressure";
  input Modelica.SIunits.Temperature T "Temperature";
  input Modelica.SIunits.Pressure delp
    "Iteration converged if (p-pre(p) < delp)";
  output Modelica.SIunits.Density d "Density";

protected 
  constant Real T_breaks[:]=Modelica.Media.R134a.R134aData.Tbreaks
    "Grid points of reduced temperature";
  constant Real dlt_coef[:, 4]=Modelica.Media.R134a.R134aData.dltcoef
    "Coefficients of cubic spline for rho_liq(T)";
  constant Real dvt_coef[:, 4]=Modelica.Media.R134a.R134aData.dvtcoef
    "Coefficients of cubic spline for rho_vap(T)";

  Boolean liquid "Is liquid";
  Boolean supercritical "Is supercritcal";
  Integer int "Interval number";
  Real Tred "Reduced temperature";
  Real localx "Oordinate of local spline";
  Integer i "Newton iteration number";
  Real dp "Pressure difference";
  Modelica.SIunits.Density deld "Density step";
  Modelica.Media.Common.HelmholtzDerivs f
    "Dimensionless Helmholtz function and derivatives w.r.t. delta and tau";
  Modelica.Media.Common.NewtonDerivatives_pT nDerivs
    "Derivatives needed in Newton iteration";
  Boolean found "Flag for iteration success";
  Integer error "1 if did not converged";

algorithm 
  Modelica.Media.R134a.R134a_ph.phaseBoundaryAssert(p, T);
  i := 0;
  error := 0;
  found := false;
  Tred := T/Modelica.Media.R134a.R134aData.data.FTCRIT;
  (int,error) := Modelica.Media.R134a.Common.FindInterval(Tred, T_breaks);
  localx := Tred - T_breaks[int];
  // set decent initial guesses for d and T
  supercritical := p > Modelica.Media.R134a.R134aData.data.FPCRIT;
  if supercritical then
    // iteration seems to work better if coming from high densities
    d := Modelica.Media.R134a.R134aData.data.FDCRIT*3.0;
  else
    liquid := T <= Modelica.Media.R134a.R134a_ph.saturationTemperature(p);
    if liquid then
      d := Modelica.Media.R134a.R134aData.data.FDCRIT*
        Modelica.Media.R134a.Common.CubicSplineEval(localx, dlt_coef[int, 1:4])*1.02;
    else
      d := Modelica.Media.R134a.R134aData.data.FDCRIT*
        Modelica.Media.R134a.Common.CubicSplineEval(localx, dvt_coef[int, 1:4])*0.95;
    end if;
  end if;

  while ((i < 100) and not found) loop
    f := Modelica.Media.R134a.R134a_ph.f_R134a(d, T);
    nDerivs := Modelica.Media.Common.Helmholtz_pT(f);
    dp := nDerivs.p - p;
    if (abs(dp) <= delp) then
      found := true;
    end if;
    deld := dp/nDerivs.pd;
    d := d - deld;
    i := i + 1;
  end while;

beutlich, 2021-01-06:
Thanks. Here as real diff for MSL 4:

```diff
--- a/Modelica/Media/R134a.mo
+++ b/Modelica/Media/R134a.mo
@@ -2471,18 +2471,18 @@ This function integrates the derivative of temperature w.r.t. time in order to a
       output SI.Density d "Density";
 
     protected
-      constant Real p_breaks[:]=R134aData.pbreaks
-        "Grid points of reduced pressure";
-      constant Real dl_coef[:, 4]=R134aData.dlcoef
-        "Coefficients of cubic spline for rho_liq(p)";
-      constant Real dv_coef[:, 4]=R134aData.dvcoef
-        "Coefficients of cubic spline for rho_vap(p)";
+      constant Real T_breaks[:]=R134aData.Tbreaks
+        "Grid points of reduced temperature";
+      constant Real dlt_coef[:, 4]=R134aData.dltcoef
+        "Coefficients of cubic spline for rho_liq(T)";
+      constant Real dvt_coef[:, 4]=R134aData.dvtcoef
+        "Coefficients of cubic spline for rho_vap(T)";
 
       Boolean liquid "Is liquid";
       Boolean supercritical "Is supercritcal";
       Integer int "Interval number";
-      Real pred "Reduced pressure";
-      Real localx "Abscissa of local spline";
+      Real Tred "Reduced temperature";
+      Real localx "Oordinate of local spline";
       Integer i "Newton iteration number";
       Real dp "Pressure difference";
       SI.Density deld "Density step";
@@ -2499,8 +2499,8 @@ This function integrates the derivative of temperature w.r.t. time in order to a
       error := 0;
       found := false;
       pred := p/R134aData.data.FPCRIT;
-      (int,error) := Common.FindInterval(pred, p_breaks);
-      localx := pred - p_breaks[int];
+      (int,error) := Common.FindInterval(Tred, T_breaks);
+      localx := Tred - T_breaks[int];
       // set decent initial guesses for d and T
       supercritical := p > R134aData.data.FPCRIT;
       if supercritical then
@@ -2509,10 +2509,10 @@ This function integrates the derivative of temperature w.r.t. time in order to a
       else
         liquid := T <= Modelica.Media.R134a.R134a_ph.saturationTemperature(p);
         if liquid then
-          d := R134aData.data.FDCRIT*Common.CubicSplineEval(localx, dl_coef[int,
+          d := R134aData.data.FDCRIT*Common.CubicSplineEval(localx, dlt_coef[int,
             1:4])*1.02;
         else
-          d := R134aData.data.FDCRIT*Common.CubicSplineEval(localx, dv_coef[int,
+          d := R134aData.data.FDCRIT*Common.CubicSplineEval(localx, dvt_coef[int,
             1:4])*0.95;
         end if;
       end if;
```

beutlich, 2021-02-06:
@henrikt-ma Any chance to review/approve #3704 which fixes your issue? Thanks.

henrikt-ma, 2021-02-07:
> @henrikt-ma Any chance to review/approve #3704 which fixes your issue? Thanks.

Done.

casella, 2025-04-01:
Unfortunately, during regression testing with other libraries, @NBrandtTLK discovered that PR #3704 caused a regression: now the setState_pTX() functions gives completely bogus values as soon as T > 370 K, as demonstrated by this simple test model

```modelica
model TestR134a
  Real T = 300 + 100*time;
  Real h = Modelica.Media.R134a.R134a_ph.specificEnthalpy(Modelica.Media.R134a.R134a_ph.setState_pTX(1e5,T));
equation
  assert(h > 3e5 and h < 1e6, "Incorrect value of enthalpy", AssertionLevel.warning);
annotation(uses(Modelica(version="4.1.0")));
end TestR134a;```

This is also against the claim of the [documentation](https://doc.modelica.org/Modelica%204.0.0/Resources/helpDymola/Modelica_Media_R134a_R134a_ph.html) that the valid range for this medium model is 0.0039 bar ≤ p ≤ 700 bar and 169.85 K ≤ T ≤ 455 K.

@DagBruck rightly pointed out during today's MAP-Lib meeting that we should not get regressions into 4.1.0, even though they succeed in fixing known issues. We then need to revert #3704 both on master and on maint/4.1.x.

Someone at TLK should look into the issue and propose solutions. If they arrive in time for the release candidate, we could still accept them in MSL 4.1.0, otherwise they will need to be postponed to 4.1.1 or 4.2.0.

casella, 2025-04-11:
This ticket won't be fixed in 4.1.0, I'll retarget it for the next release

casella, 2025-04-15:
This will also be fixed by #4626, which is to be back-ported to maint/4.1.x.

All's well that ends well 😃.

PR #3704 (2021-04-09): Changed dofpT issue #3695
Changed code for function dofpT for testing to resolve #3695.
closed 2025-04-16 by Esther-Devakirubai