modelica/ModelicaStandardLibrary #3711: New spikes in RectifierCenterTap2mPulse reference results

henrikt-ma, 2021-01-12:
The 3.2.3 and 4.0.0 reference results for these models differ due to spikes in `rectifier.thyristor_p.idealThyristor[3].off` appearing in the 4.0.0 versions:
- `Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_R`
- `Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_RL`

In both versions, there's a single spike at time 1.667e-3:
[image]

When simulated using System Modeler, I get the same result as in the 3.2.3 versions (that is, without those spikes), which makes me wonder if those spikes really are a new and expected feature?

henrikt-ma, 2021-01-12:
The spike is also seen for this model:
- `Modelica.Electrical.PowerConverters.Examples.ACDC.RectifierCenterTap2mPulse.ThyristorCenterTap2mPulse_RLV`

AHaumer, 2021-01-14:
Yes these spikes are new, and no this is not an expected feature. 
To be honest, normally I'm not looking at thyristor_x[n].off since this is only an auxilliary variable. The only reason for not making it protected is to enable proper initialization of the thyristors. These spikes don't affect the results an engineer is looking at, i.e. current and voltages. 
It stems from infinitesimal numerical inaccuracies. For 4.0.0 I've made PowerConverters.ACDC.Control.Signal2mPulse easier to read. Here we start a timer at a zero crossing of the corresponding voltage, then comparing the timer with respect to the firing angle. The difference is:
```
3.2.3 timer.y > firingAngle.y/(2*pi*f)
4.0.0 timer.y*2*f > firingAngle.y/pi
```
We could switch back to the old implementation, but I think it isn't worth to do so.

henrikt-ma, 2021-01-14:
How about simply excluding the `thyristor_x[n].off` from the comparison signals?

AHaumer, 2021-01-14:
I don't know why these signal are in the ComparisonSignals.txt ... I'll create a PR.
Is it ok to cerate the PR for both #3711 and #3713 ?
Both have the same bugfix, both are for PowerConverters,

maltelenz, 2021-01-25:
@AHaumer I see no problem with a single PR to remove comparison signals for both issues.

maltelenz, 2024-02-29:
> I don't know why these signal are in the ComparisonSignals.txt ... I'll create a PR.

@AHaumer It seems the PR hasn't been created, maybe now is a good time, so we can get the fixes for the reference results in 4.1.0?