modelica/ModelicaStandardLibrary #3713: Ill-conditioned result variable in Dimmer_RL

henrikt-ma, 2021-01-13:
This concerns the selection of comparison variables for the example:
- `Modelica.Electrical.PowerConverters.Examples.ACAC.Dimmer_RL`

I suspect that `vH01.y_arg` has an ill-conditioned result.  It is the result of an `atan2` computation in `Modelica.Blocks.Math.RectangularToPolar`:
```
  y_arg = Modelica.Math.atan2(u_im, u_re);
```
The result variables corresponding to the two arguments to `atan2` are `vH01.mean2.y_last` and `vH01.mean1.y_last`, respectively:
[image]

Before time 1 and after time 7, there are periods where both variables look like numeric noise around zero when simulated using System Modeler.  The reason I'm tempted to trust our result is that the `atan2` result matches the reference result anywhere the variables are significantly non-zero.

Can someone please confirm that the two variables are always sufficiently non-zero to make the `atan2` well defined?

AHaumer, 2021-01-14:
@henrikt-ma No I can't confirm that the two variables are always sufficiently non-zero to make the atan2 well defined. 
It's the old problem of converting cartesian coordinates to polar coordinates in periods where real part and imaginary parts are small, but engineers tend to look at polar represenation in that cases and normally they ignore the angle when the length is sufficiently small. BTW, it's the same for iH01.y_arg. 
It's dangerous to change the implementation of Modelica.Blocks.Math.RectangularToPolar since it's used in many other models. 
Do you have a suggestion for converting cartesian coordinates to polar coordinates that is robust and stable even in difficult situations, i.e. real and imaginary part near zero, or even a phasor (x, y) with constant angle but length first decreasing towards zero and the increasing in the opposite direction (does the angle jump by pi or the sign of the length)?
Is it acceptable to remove such tricky post-proccessing signals from comparisonSignals,txt, keeping mainly continuos signals?

henrikt-ma, 2021-01-14:
> @henrikt-ma No I can't confirm that the two variables are always sufficiently non-zero to make the atan2 well defined.
> It's the old problem of converting cartesian coordinates to polar coordinates [...]

Right, I missed that one because System Modeler happens to get a small but non-zero value for `iH01.mean2.y_last` at times where `iH01.mean1.y_last` is zero.  The resulting angle we get matches the reference result, for what it's worth…

> It's dangerous to change the implementation of Modelica.Blocks.Math.RectangularToPolar since it's used in many other models.

Yes, I definitely wouldn't suggest doing that!

> Do you have a suggestion for converting cartesian coordinates to polar coordinates that is robust and stable even in difficult situations, i.e. real and imaginary part near zero, or even a phasor [...]

No, I think it's just the nature of polar representation that the angle isn't well defined at the origin.

> Is it acceptable to remove such tricky post-proccessing signals from comparisonSignals,txt, keeping mainly continuous signals?

Yes, and I don't see why one wouldn't do comparison of the cartesian coordinates instead.

AHaumer, 2021-01-14:
> Yes, and I don't see why one wouldn't do comparison of the cartesian coordinates instead.
Because an electrical engineer is interested in the amplitude (and sometimes the phase angle) of the harmonics,

henrikt-ma, 2021-01-14:
> > Yes, and I don't see why one wouldn't do comparison of the cartesian coordinates instead.
> 
> Because an electrical engineer is interested in the amplitude (and sometimes the phase angle) of the harmonics,

So then these are the variables you should include in a figure annotation.  It doesn't mean the variables are suitable as comparison signals.  (Ideally, comparison signals should be a subset of variables shown in figures, so that there is reason to believe that someone is actually looking at the results of the compared variables.)