modelica/ModelicaStandardLibrary #3726: Documenting inputs in models and blocks

eshmoylova, 2021-01-18:
In `Modelica.Mechanics.MultiBody.Parts.FixedTranslation` (for example) there are two inputs `specularCoefficient` and `color`. Should those inputs be documented as parameters? Looking at the help pages generated for it (for MSL 3.2.3, for example) in different tools, those two inputs are reported by one tool as [parameters](https://doc.modelica.org/Modelica%203.2.3/Resources/helpDymola/Modelica_Mechanics_MultiBody_Parts.html#Modelica.Mechanics.MultiBody.Parts.FixedTranslation), by another as [inputs](https://doc.modelica.org/Modelica%203.2.3/Resources/helpWSM/Modelica/Modelica.Mechanics.MultiBody.Parts.FixedTranslation.html) (as I would expect), and are [omitted](https://doc.modelica.org/Modelica%203.2.3/Resources/helpMapleSim/Mechanics/MultiBody/Parts/index.html#FixedTranslation) altogether by yet another.

How should they be reported? The user needs to be able to specify a modification for an input variable which usually done through a parameter inspector/setter dialog in the tools. But it is misleading to call them parameters because their values can change during the simulation. How should they be presented to a user?

I am not sure if this is an issue with the library or with the Specification. Since I had a concrete example with the library class decided to post it here.

beutlich, 2021-01-18:
I believe you are referring to the Dymola heuristics to add any non-constant component having a Dialog annotation to the auto-generated table of class parameters (even though inputs, outputs or ordinary variables cannot be considered as constant parameters in the strict sense of Modelica language).

These additional and auto-generated documentation is not defined in the Modelica language specification and therefore its assumptions were removed by https://github.com/modelica/ModelicaStandardLibrary/pull/2501/commits/d4aca424b8f12a6ea2ad030fda507f39768da9fc (squashed as f2eec0eff074fe7e20a13593f8c9e07b8a20dd6f) for MultiBody.Visualizers along with #2501 for MSL v4.0.0.

I propose to close this issue as tool-issue.

maltelenz, 2021-01-19:
It is not immediately obvious to me why we have empty Dialog annotations in the library, for example here:
https://github.com/modelica/ModelicaStandardLibrary/blob/403e7c1e9c9399fe0481e593815deb4bae094203/Modelica/Mechanics/MultiBody/Visualizers/Advanced/Arrow.mo#L21
I can't find anything in the specification saying that this means anything, and therefore, I would consider it bad style to have empty annotations present.

If it does or should mean something, that should be in the specification. The standard library should not play into one tools heuristics (if this is what this is about).

eshmoylova, 2021-01-19:
Before closing this issue, I just want to clarify a couple of things. As far as I can see, d4aca42 simply documents those inputs as such, right? However, if you look at classes defined in `Modelica/Mechanics/MultiBody/Parts` the following ones also have inputs that have some Dialog annotation and would also be potentially listed as parameters by some tools:
* `Body`
* `BodyBox`
* `BodyCylinder`
* `Fixed`
* `FixedRotation`
* `FixedTranslation`
* `PointMass`
* `Rotor1D`
I do not see any clarifications about the meaning of inputs in the Documentation annotations of those classes. There might be others I just took a look at one folder only.

eshmoylova, 2021-01-19:
> It is not immediately obvious to me why we have empty Dialog annotations in the library, for example here: [...]

I think it means that the default arguments are used, the same as for a record constructor with empty argument list. The annotations are defined as records in the specification and in case of the Dialog annotation there are default arguments defined by the specification for all fields:

> Annotation Dialog is defined as:
> ```
> record Dialog
>   parameter String tab = "General";
>   parameter String group = "Parameters";
>   parameter Boolean enable = true;
>   parameter Boolean showStartAttribute = false;
>   parameter Boolean colorSelector = false;
>   parameter Selector loadSelector;
>   parameter Selector saveSelector;
>   parameter String groupImage = "";
>   parameter Boolean connectorSizing = false;
> end Dialog;
> record Selector
>   parameter String filter = "";
>   parameter String caption = "";
> end Selector;
> ```

maltelenz, 2021-01-19:
> I think it means that the default arguments are used, the same as for a record constructor with empty argument list.

Interesting take. Maybe this could be made clearer in the specification if that is the intended interpretation. I would not have arrived at this conclusion by reading the current specification text.

henrikt-ma, 2021-01-20:
This reminds me of https://github.com/modelica/ModelicaSpecification/issues/2476, which has been settled and merged, but then appears to have been lost again. @HansOlsson, this looks like something in need of a 3.5 hotfix!

henrikt-ma, 2021-01-20:
> Interesting take. Maybe this could be made clearer in the specification if that is the intended interpretation. I would not have arrived at this conclusion by reading the current specification text.

This is also how I interpret it, and seems like an underlying assumption in modelica/ModelicaSpecification#2476.

The thing that I believe might deserve some sort of general comment in the specification is that there can be a difference between providing an annotation with only defaults, and not providing the annotation at all.

henrikt-ma, 2021-01-22:
The default value seen for `group` in https://github.com/modelica/ModelicaStandardLibrary/issues/3726#issuecomment-762911688 is actually not according to what has been decided.

I have set up a PR for addressing this and general cleanup of the description `Dialog`: #2819

eshmoylova, 2021-02-12:
With MO#2819 merged I think the issue can be closed. @maltelenz do you want to open a separate issue against the specification to clarify the default values for the annotations?

closed 2021-02-12 by eshmoylova

maltelenz, 2021-02-15:
@eshmoylova Opened new issue in #3749, thanks for the reminder.