modelica/ModelicaStandardLibrary #3727: Value of motor.uniformNoise.localSeed in ActuatorWithNoise

henrikt-ma, 2021-01-19:
After lots of struggle trying to reproduce the reference result for `Modelica.Blocks.Examples.NoiseExamples.ActuatorWithNoise`, me and @maltelenz have narrowed it down to the computation of `motor.uniformNoise.localSeed`.  By setting the value to -1385457713, as described [here](https://github.com/modelica/ModelicaStandardLibrary/pull/3434/files), we're able to reproduce the reference result.  However, this doesn't look like the right value.

Please help us understand why the value 188641931 computed by this test program is not the correct value:
```
#include <string.h>
#include <stdio.h>

/* AP hash function macro variant of the one listed at
   http://www.partow.net/programming/hashfunctions/index.html#APHashFunction

   Copyright (C) 2002, Arash Partow

   Permission is hereby granted, free of charge, to any person obtaining a copy
   of this software and associated documentation files (the "Software"), to deal
   in the Software without restriction, including without limitation the rights
   to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
   copies of the Software, and to permit persons to whom the Software is
   furnished to do so, subject to the following conditions:

   The above copyright notice and this permission notice shall be included in all
   copies or substantial portions of the Software.

   THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
   IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
   FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
   AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
   LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
   OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
   SOFTWARE.
*/
#define HASH_AP(key, keylen, hash) \
do { \
    unsigned _hb_keylen = (unsigned)keylen; \
    const unsigned char *_hb_key = (const unsigned char*)(key); \
    unsigned int i; \
    hash = 0xAAAAAAAA; \
    for (i = 0; i < _hb_keylen; _hb_key++, i++) { \
        hash ^= ((i & 1) == 0) ? (  (hash <<  7) ^ (*_hb_key) * (hash >> 3)) : \
                                 (~((hash << 11) + ((*_hb_key) ^ (hash >> 5)))); \
    } \
} while (0)

int
main()
{
  union hash_tag {
    unsigned int iu;
    int          is;
  } h;
  const char instanceName[] = "ActuatorWithNoise.motor.uniformNoise";
  size_t len = strlen(instanceName);
  HASH_AP(instanceName, len, h.iu);
  printf("HASH_AP(\"%s\"): %d\n", instanceName, h.is);
  return 0;
}
```

beutlich, 2021-01-19:
Looks like a duplicate issue as reported by https://github.com/modelica/ModelicaStandardLibrary/issues/3415#issuecomment-586456274 for Modelica.Utilities.Strings.hashString, which works/worked differently in Dymola due to merged compilation modules.

We should set a fixed local seed for that exampel model.

henrikt-ma, 2021-01-19:
Yes, must be the same problem.  It also affects `Modelica.Blocks.Examples.Noise.DrydenContinuousTurbulence`.

beutlich, 2021-01-19:
It also is addressed by the next release of uthash where we can fix it. See https://github.com/troydhanson/uthash/commit/053bed12453660f2e2f7733438fc9312bc54adb5 for now.

henrikt-ma, 2021-01-19:
> It also is addressed by the next release of uthash where we can fix it. See [troydhanson/uthash@053bed1](https://github.com/troydhanson/uthash/commit/053bed12453660f2e2f7733438fc9312bc54adb5) for now.

I have no idea how this would fix Dymola, but if you say so…

beutlich, 2021-01-19:
> 
> 
> > It also is addressed by the next release of uthash where we can fix it. See [troydhanson/uthash@053bed1](https://github.com/troydhanson/uthash/commit/053bed12453660f2e2f7733438fc9312bc54adb5) for now.
> 
> I have no idea how this would fix Dymola, but if you say so…

There you go: #3728

henrikt-ma, 2021-01-20:
I'm happy as long as the models aren't modified to use a fixed seed just because that's the most convenient way to obtain a correct reference result.  I see two better approaches to getting a correct reference result:
- Use another tool.
- Temporarily modify the models while computing the reference result, but never commit those changes.

beutlich, 2021-01-20:
Do you insist on having these corrected ref results now? Otherwise it will be simply corrected with the next MSL release after #3728 is merged. Additionally, there are more questionable ref results you reported.

henrikt-ma, 2021-01-20:
> Do you insist on having these corrected ref results now? Otherwise it will be simply corrected with the next MSL release after #3728 is merged. Additionally, there are more questionable ref results you reported.

That's a question for @maltelenz.  For me, just don't remove the use of automatic local seed, because I think it is an important use case that should remain the seeding method of choice for basic examples.

maltelenz, 2021-01-20:
In an ideal world, there would be new reference results in cases like this (where the old one is deemed incorrect/questionable), without having to wait for a new release of MSL. The effect of not having that, means that no tools can test the models in question against their tool, which is obviously not great (at least for tool makers).

That said, I think we (System Modeler) can live with the current state of affairs if needed, lowering our test ambition for models such as these until the next version of MSL is released.

PR #3728 (2021-02-26): Fix Modelica.Utilities.Strings.hashString to always use provided HASH_AP
This closes #3727 by making ModelicaStrings.c independent of the inclusion order. It also reverts #3442 by reintroducing the test for exact values.

This PR uses a non-released version of uthash. I will update to uthash v2.3.0 whenever it is available.

@Quuxplusone FYI
closed 2021-02-26 by beutlich