modelica/ModelicaStandardLibrary #3734: Primary and secondary vs. side 1 and 2

christiankral, 2021-01-24:
The transformer models provided in the MSL rely on the following wording:

- Index 1: **primary**
- Index 2: **secondary**

[image]

This is unfortunately not correct. According to Electropedia the following wording applies:

- [Primary winding](http://www.electropedia.org/iev/iev.nsf/display?openform&ievref=421-03-06): a winding which, in service, receives active power from the supply network
- [Secondary winding](http://www.electropedia.org/iev/iev.nsf/display?openform&ievref=421-03-07): a winding which, in service, delivers active power to the load circuit

So obviously the terms primary and secondary refer to the power flow of a transformer. Therefore, the primary winding becomes secondary winding if the power flow direction changes. Like it or not (I don't like this definition), this is how it is used and how we shall use it in the MSL, too.

In order to be in line with the wording of Electropedia we could strictly stick with the wording 
- Side 1
- Side 2

christiankral, 2021-01-24:
The wording "side 1" and "side 2" is already partly applied in #3697

dietmarw, 2021-03-09:
I have no strong opinion here. For me both definition work.